FreeOpcUa/python-opcua #868: Performance Optimizations for Large Quantity of Tags

russellkir, 2019-07-24:
Hi,
I have the following sample code I use in a microservice to connect to an OPC UA server and pull down tags via their respective node values:
`from opcua import Client
import time

# example address to server
conn = Client("opc.tcp://192.168.222.100:49320")
conn.connect()

# example nodes in list, typically much larger
node_list = ["ns=2;s=Simulation Examples.Functions.Ramp1", "ns=2;s=Simulation Examples.Functions.Ramp10"]

t1 = time.time()
for n in node_list:
    node = conn.get_node(n)
    value = node.get_value()
    print("Node: {}, Value: {}".format(n, value))
t2 = time.time()
print("Time taken to query {} nodes: {}".format(len(node_list), round(abs(t2-t1), 3)))
`
After trying this out and looking at the docs a bit, I have a few questions about my implementation and was hoping for some brief guidance :)

1. Is there a way to pass the list of nodes to the Client and get back all values at once?  I think this would greatly optimize the overall response time, as in some cases, I'm querying hundreds if not thousands of nodes.
2. Is there anything I can do to be more performant with the library?  The logic shown above runs at 1000ms polling intervals and I'm trying to keep up with a Kepware server on our network that seems to be able to handle even 5000 nodes no problem.  In the worst case scenario, I might see 10 second latency for 100+ nodes.

Other than that, if there is just anything glaringly obvious I'm missing here that I should consider it would be great to get your thoughts on this.

Thank you for your time!

zerox1212, 2019-07-25:
As has been posted many times on here, use a subscription if you want the best performance.

If you want to read many nodes at once you have to write custom lower level UA functions.
https://github.com/FreeOpcUa/python-opcua/issues/604#issuecomment-398054448

russellkir, 2019-07-25:
@zerox1212 got it, thank you for the suggestions.

oroulet, 2019-08-01:
for reference, there is now a Client.get_value and Client.set_values methods to read and write on  many nodes at onces

closed 2019-08-01 by oroulet