FreeOpcUa/python-opcua #873: Why subscription is faster than get_value

EOFFJM, 2019-08-05:
I would like to know why subscription is faster than using get_value function. Is it because get_value uses request-response communication and subscription used pub-sub?

destogl, 2019-08-05:
What do you exactly mean by faster? How did you measure it?

schroeder-, 2019-08-05:
get_value triggers a communication per call. Subscriptions trigger a communication once to register and then gets updated in request-response communication every time the value changes (If not using filters). You can get better performance with get_value if you bundle your read requests (**get_values**), this reduces the amount of communications a lot.

EOFFJM, 2019-08-05:
> What do you exactly mean by faster? How did you measure it?

I added some code in the client program that would measure the time it takes to retrieve node values from the server.

EOFFJM, 2019-08-05:
> get_value triggers a communication per call. Subscriptions trigger a communication once to register and then gets updated in request-response communication every time the value changes (If not using [...]

Thanks.

destogl, 2019-08-12:
@EOFFJM if you are satisfied with the answer, can you close the issue?

EOFFJM, 2019-08-21:
> get_value triggers a communication per call. Subscriptions trigger a communication once to register and then gets updated in request-response communication every time the value changes (If not using [...]

Hi thanks for the answer. How do you know this? Is there some documentation that explains this?

zerox1212, 2019-08-21:
You can read the OPC UA spec about how the subscription service works.

It's faster because the data from a subscription update has much less overhead data than a get_value request. You should be able to see this by inspecting the get_value vs. subscription packets with wire shark.

dudanisu, 2019-09-04:
Can you please give a example of get_values()? How can I pass the variables to get_values() and also how can I get the list of values? Thanks.

EOFFJM, 2019-09-07:
> You can read the OPC UA spec about how the subscription service works.
> 
> It's faster because the data from a subscription update has much less overhead data than a get_value request. You should be able to see this by inspecting the get_value vs. subscription packets with wire shark.

Is there any document about the get_value function? I cannot find it.

EOFFJM, 2019-10-02:
> get_value triggers a communication per call. Subscriptions trigger a communication once to register and then gets updated in request-response communication every time the value changes (If not using [...]

Is there any documentation or something that explains about the get_value function?

EOFFJM, 2019-10-02:
> You can read the OPC UA spec about how the subscription service works.
> 
> It's faster because the data from a subscription update has much less overhead data than a get_value request. You should be able to see this by inspecting the get_value vs. subscription packets with wire shark.

Thanks. So get_value request is request/response? And subscription is publish/subscribe communication?

closed 2019-10-23 by EOFFJM