FreeOpcUa/python-opcua #874: Attribute Reading for non-base types

GalaxyGorilla, 2019-08-07:
Hey, I have a general question about how the client does node attribute reading. I wiresharked the read attribute request (from the GUI client, if that matters) against the server and was slightly suprised by what I got back as a response. So here is the element from the `NodesToRead` within a `ReadRequest`:

[image]

So here we request the `NodeClass` (which is an enumeration). And this is what I get back from the server:

[image]

So, why is that weird? For me it looks like you return the enumeration value (which is `1`) directly without encoding it as `NodeClass` datatype, e.g. as proper enumeration. So what I had expected is rather an extension object containing the encoded `NodeClass` (with enumeration value `1`). Is this a misunderstanding in my view on OPC UA or is this a bug within your stack?

oroulet, 2019-08-07:
you could do the test with another stack to check. I do not think nodeclass can be packed into an extension object so I do not think we can return something else than an int32 in that case, but maybe VariantType could be something else... I do not remember

zerox1212, 2019-08-07:
From what I remember, extension objects don't pack anything about their own structure. This is exactly why the client must get and register extension objects from the server before they can be handled.

GalaxyGorilla, 2019-08-07:
From the spec part 6:
```
An ExtensionObject is a container for any Structured DataTypes which cannot be encoded as one of the other built-in data types. The ExtensionObject contains a complex value serialized as a sequence of bytes or as an XML element. It also contains an identifier which indicates what data it contains and how it is encoded.
Structured DataTypes are represented in a Server address space as sub-types of the Structure DataType.
```
So since enums are no structures (no subtype of node 22) they cannot be packed into an extension object, alright. But in general I think you can put in there whatever you want as long as it is a structure. What a bloody mess, why is there a datatype `NodeClass` if you can't use it? It is even present in the old .bsd file holding the schemas for decoding.

oroulet, 2019-08-07:
Yes ua is very strange at several places... To write it politely

oroulet, 2019-08-07:
The worst is well structures(extension objects) and custom types

GalaxyGorilla, 2019-08-07:
FYI, even though this is probably not public yet:

https://opcfoundation.org/forum/opc-ua-implementation-stacks-tools-and-samples/variants-and-enumerations/#p1902

zerox1212, 2019-08-07:
What you posted is close to what I remember. The only thing that gets packed is the "identifier" which the client must already know about.

GalaxyGorilla, 2019-08-09:
I close this, for further infos see link above.

closed 2019-08-09 by GalaxyGorilla

zerox1212, 2019-08-09:
What he says is true. Enumerations and extension objects were definitely "inspired" by C style language, which is annoying.