FreeOpcUa/python-opcua #875: Trouble connecting using python-opcua but Prosys client works

achintya-kumar, 2019-08-07:
client = Client("opc.tcp://10.80.0.38:4840")
client.connect()
Received an error: MessageAbort(error:StatusCode(Good), reason:The outgoing message including all chunks is too large.)
Received an error: MessageAbort(error:StatusCode(Good), reason:The outgoing message including all chunks is too large.)
Protocol Error
Traceback (most recent call last):
  File "/home/kumar/IdeaProjects/umati-emo/venv/lib/python3.6/site-packages/opcua/client/ua_client.py", line 96, in _run
    self._receive()
  File "/home/kumar/IdeaProjects/umati-emo/venv/lib/python3.6/site-packages/opcua/client/ua_client.py", line 114, in _receive
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "/home/kumar/IdeaProjects/umati-emo/venv/lib/python3.6/site-packages/opcua/client/ua_client.py", line 124, in _call_callback
    .format(request_id, self._callbackmap.keys())
opcua.ua.uaerrors._base.UaError: No future object found for request: 0, callbacks in list are dict_keys([2])
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/home/kumar/IdeaProjects/umati-emo/venv/lib/python3.6/site-packages/opcua/client/client.py", line 259, in connect
    self.create_session()
  File "/home/kumar/IdeaProjects/umati-emo/venv/lib/python3.6/site-packages/opcua/client/client.py", line 360, in create_session

oroulet, 2019-08-14:
what server?

achintya-kumar, 2019-08-17:
Ok, so I've resolved this issue. It was related to max response message size. I increased the value to 65535 from 0 and it works fine now. Thanks!

closed 2019-08-17 by achintya-kumar