FreeOpcUa/python-opcua #879: Node creation at running server

Lukas1811, 2019-08-29:
Currently I'm trying to create multiple instances of a custom object type on a running server (creation is triggered by a method of another object). The other device is created before I call server.start(), but when I use the same function at the running server, the creation won't work but I don't get any errors. Does anyone have an idea how I could create object instances at the running server.

I use the FreeOpcUa GUI client the trigger the methods and watch the device structure, I hope it's not the client having a problem, because I had a lot of problems with the modeler too.

zerox1212, 2019-08-29:
How are you creating the instances? With `instantiate()` function?

If you don't trust the client then try UA Expert.

Lukas1811, 2019-08-30:
I do the node creation like this
``` python
self.object = self.server.root_node.add_object(self.namespaceIndex, self.name , self.typeId)
```

closed 2019-09-03 by Lukas1811