FreeOpcUa/python-opcua #880: PubSub support

GalaxyGorilla, 2019-08-30:
Hey, I'm using the client and server to do some research on OPC UA and want to checkout how pubsub is working. Do you have any support for pubsub? Looking into the examples I don't see anything but the docu looks like there is actually some support for this.

Maybe I overlooked an example or searched for the wrong keywords? Maybe there is some good example floating around on github?

oroulet, 2019-08-30:
You mean subscription? Or the new udp based protocol? The udp stuff is not implemented. Patch welcome

-------- Original Message --------

GalaxyGorilla, 2019-08-30:
I mean subscriptions, but in the sense of PubSub (part 14 of the specs). Don't know if that stuff is new ... but yea it is UDP based when you go brokerless. Or you go with AMQP or MQTT (both use iirc TCP) when a broker is involved.

oroulet, 2019-08-30:
Part 14 is not implemented. Patch welcome

-------- Original Message --------

mila-mg, 2021-02-03:
Hello! I am fairly new to python and definitely a beginner in terms of opcua client installation. However, I am trying to create a client that is subscribed to a server running machine system values and want the client to be notified whenever specific values change.

My first question is: is this possible with python opcua? From what I understood, this has not been implemented yet?

My second question would be: what exactly do you mean when saying: "EDIT: sorry I think I mixed things up in the docu with the subscription service (which has nothing to do with pubsub)"? What is the difference between subscription service and publish-subscribe set up?

I am grateful for every help!

closed 2021-02-03 by AndreasHeine