FreeOpcUa/python-opcua #881: [HELP] How to add locale into LocalizedText ?

YiKai07, 2019-09-02:
Hi, 
I have been looking for a function to add local into LocalizedText, but I only found localid in OPC specification part 3, and I also try to add 
```
<ListOfLocalizedText>
        <LocalizedText>
            <locale>en</locale>
            <Text>Running</Text>  
        </LocalizedText>
        .....
</ListOfLocalizedText>
```
into my xml file and import into my code, but it still dosen't work.

and I also try this code found in the example:
```
def enum_to_stringlist(a_enum):
    items = []
    for value in a_enum:
        items.append(ua.LocalizedText(value.name))
    return items
```
but ua.LocalizedText only take one value, and if I gave a String like "en, Running", it will output like 
`"", "en, Running"`

oroulet, 2019-09-02:
As written in readme, this is one of the things not implemented. But this should probably be easy. It is just that I had (and still have) absolutely no idea about what people expect from this locale stuff. How do you think this should work?

-------- Original Message --------

YiKai07, 2019-09-02:
I'm still looking for solution, so I've no idea how to work this either.
seems like add Locale from xml file it's worked, but I haven't get the correct output yet.

maxim-s-barabash, 2019-09-02:
The OPC UA client must explicitly provide the OPC UA specific language "LocaleId" to change the alarm texts.  UaExpert provided the OPC UA specific "LocaleId" as value for the parameter "General.LocaleId".

On server side process ua.ActivateSessionParameters.LocaleIds 

...