FreeOpcUa/python-opcua #882: Nodeset import doesn't load variables

Lukas1811, 2019-09-03:
If you import a nodeset from a XML file, the XmlImporter used by the import_xml function of the server, doesn't load the variables as subtype of custom datatypes.
In my case I created a nodeset in a nodeset modeler (DI namespace, custom namespace, custom events, data types and object types) and imported it to my server. I don't get any errors, but if I watch the server in a client, the data types are imported, but the variables aren't imported.
I don't know if it's an user error but I don't think so, because the [opcua modeler](https://github.com/FreeOpcUa/opcua-modeler) has the same problem.

maxim-s-barabash, 2019-09-03:
Hi. show the minimum nodeset with comments what is expected

Lukas1811, 2019-09-03:
An example datatype in my nodeset looks like this 
```xml
<opc:StructuredType BaseType="uaExtensionObject" Name="AddrDataType">
    <opc:Field Name="Addr" TypeName="opc:String" />
    <opc:Field Name="AddrType" TypeName="opc:String" />
</opc:StructuredType>
```
I import it like this :

```python
def __load_namespaces(self, namespaces: list) -> bool:
    for namespace in namespaces:
       self.import_xml(namespace)
       self.namespaces = self.get_namespace_array()
       log.info("Import namespace from file {0} successfull !".format(namespace))
```
I pass an array with the paths to the official DI and custom nodeset. After this I start the server.

maxim-s-barabash, 2019-09-03:
> ```xml
> <UAVariable BrowseName="1:AddrDataType" DataType="String" NodeId="ns=1;i=2003" ParentNodeId="ns=1;i=2001">
> ```
> DataType="String"

Is this the correct value for a DataType?

Lukas1811, 2019-09-03:
I don't know but it's created with the opcua modeler, that also uses python-opcua, that's why I think my server should understand what to do with that data.
I think only the TypeDictionary should matter, because I don't create ab instance in the template, I only create the data type

zerox1212, 2019-09-04:
@maxim-s-barabash I think python-opcua will look up that string in the autogenerated `ObjectIds` class to get the numerical UA ID. https://raw.githubusercontent.com/FreeOpcUa/python-opcua/master/opcua/ua/object_ids.py

I'm not sure why it isn't working unfortunately. The XML stuff is very hard to follow. To be honest I never had a reason to make datatypes. To me it was easier to make custom objects and call 'instantiate()` to make instances of them.