FreeOpcUa/python-opcua #884: Issue with decoding Byte type in NodeID

smith2t, 2019-09-05:
having issues decoding the following byte to string in python using opcua:
    b'\x00\x00\x00\x00\x01\x00\x00\x00\xd1\x07\x00\x00\x00\x00YtY\x00\x00\x00'
This is an IP21 NodeID. funny thing is that .NET OPCUA clients decode this just fine.
I do know its opaque byte type that basically the app can do what it wants, but am stumped as to how the .NET clients work. I have looked through the source code but haven't seen anything helpful in pointing me to a solution.

zerox1212, 2019-09-05:
What is the python output? What is supposed to be?

smith2t, 2019-09-05:
python output is: b'\x00\x00\x00\x00\x01\x00\x00\x00\xd1\x07\x00\x00\x00\x00YtY\x00\x00\x00' which should be like AAAABBBEWEEHHHWWWW as an example. I will try to get the actual and post.

oroulet, 2019-09-05:
Where do you want the human form to appear?

smith2t, 2019-09-05:
Need to have human readable form in script output. I’m having issues figuring how to decode byte as it’s form is in known in python.

oroulet, 2019-09-05:
So what you doing is that you are reading a byte nodeid and want to encode it to utf8? This has nothing to do with opcua. Try mybytes.decode('utf-8') or maybe some windows encoding if that is your world

zerox1212, 2019-09-05:
I don't really understand where this is happening for you. Do you have an example code of what you are doing? Are you using python-opcua client or server?

smith2t, 2019-09-05:
That's just it...I don't have the options in the .NET console yet it will resolve the bytes to AAAAAAEAAAAvBgAAAABZdFkAAAA= so it knows some how resolve yet I don't what it should be decoded to and cant find any decode() in python that work. I opened this for help as it looks like the source has a ua_utils.py with some decoding but once invoked it resolved to the byte code in txt.

corydodson, 2019-09-05:
Below are screenshots from IDLE (python interface) and the OPC UA GUI interface. They are both looking at the same node and providing a different representation of the same node id:
1) python interface: ns=2;b=b'\x00\x00\x00\x00\x01\x00\x00\x00\xd1\x07\x00\x00\x00\x00YtY\x00\x00\x00'
2) GUI interface: ns=2;b=AAAAAAEAAADRBwAAAABZdFkAAAA=

The primary goal is to represent the node id as shown by the GUI from within Python. In other words, get Python to output 2) instead of 1). I have also tried the to_string() method for the node id object within Python, but it just puts double quotes around the text above (pasted here for completeness: "ns=2;b=b'\\x00\\x00\\x00\\x00\\x01\\x00\\x00\\x00\\xd1\\x07\\x00\\x00\\x00\\x00YtY\\x00\\x00\\x00'")

[image]
[image]

zerox1212, 2019-09-06:
Now I understand. That is interesting that code and GUI are different. @oroulet will probably have insights into this. If I get time I'll dig into it.

maxim-s-barabash, 2019-09-06:
```
>>> import base64
>>> base64.b64decode("AAAAAAEAAADRBwAAAABZdFkAAAA=")
'\x00\x00\x00\x00\x01\x00\x00\x00\xd1\x07\x00\x00\x00\x00YtY\x00\x00\x00'
```

oroulet, 2019-09-06:
So that guy had nodeids encoded with base64? It is not an issue, but really nothing related to opcua. Why should we display bytes nodeids as base64? Bytes are bytes

maxim-s-barabash, 2019-09-06:
opcua not should. I think so. this is a feature of the implementation of the viewer

oroulet, 2019-09-06:
but why should we decode bytes as base64 on viewer? are all bytes nodeid supposed to be base64?

oroulet, 2019-09-06:
thinking about it: the .NET viewer probably encode the bytes into base64 to display them, to avoid sending some strange codes to their  display widget. I do not think we have the same issue in python/qt so we have no reasons to encode things to base64

smith2t, 2019-09-06:
I’m sure the application GUI encoded to base64. I should have tried that. I’m not sure how to pursue this as a bug either. When opaque bytes are allowed it could have been anything. Thanks for the insight on this. I’m going to close this and hopefully will help others if this comes up again in the future.

closed 2019-09-06 by smith2t

deeTEEcee, 2022-04-09:
> but why should we decode bytes as base64 on viewer? are all bytes nodeid supposed to be base64?

Shorter and easier to read than a long series of bytes on the string. Also, the `ns=;b=<bytes>` format will not work right now so that's a bug.

dostojewskyi, 2023-09-11:
For later joiners. 
IP21 encodes NodeIDs on another base. 
muse the ByteString provided by Aspen and make an base64.decode() use this decoded bytestring for calling get_nodes() which will encode it to hex on the same base.

hih4323, 2023-10-22:
> For later joiners. IP21 encodes NodeIDs on another base. muse the ByteString provided by Aspen and make an base64.decode() use this decoded bytestring for calling get_nodes() which will encode it to hex on the same base.

Do you have a small code example of this?

Thanks in advance!