FreeOpcUa/python-opcua #889: How to link a PLC sensor tag with FreeOPCUA server?

petric3, 2019-09-12:
I would appreciate some explanation to better understand creating the FreeOPCUA server. I read "Python OPC-UA Documentation, Release 1.0", but lacks of explanation for a simple operational engineer, i.e. me :)

We are planning to replace Siemens OPC server with Python FreeOPCUA. I was able to establish FreeOPCUA client and read the signals on existing Siemens OPC server. My problem is to create FreeOPCUA server on a new Windows-based workstation and read a sensor signal from a PLC. 

Firstly, the URI consists from IP&Port, and I do not understand how Port is selected in FreeOPCUA: is there any rule or it's just randomly selected?  

Secondly, to add a signal in Siemens OPC server, we add signal address, e.g. B2 DB11.DBX3.0	X1010.3, via Siemens server Symbol editor to which more friendly name or Identifier, e.g. Temperature001,is given.  This name is than used to observe the values of the sensor via client. 

I tried to do sth similar in FreeOPCUA:
```

server = Server()
url    = "opc.tcp://10.15.11.175:55102"
server.set_endpoint(url) 

name     = "OPC01"
addspace = server.register_namespace(name)
 
node   = server.get_objects_node()
Params = node.add_object(addspace,"Parameters")

Temp1 = Params.add_variable(addspace,"B2 DB11.DBX3.0	X1010.3",0) 
Temp1.set_read_only()

server.start()

while True: #Read & display the temperature values
    
    print(Temp1.get_value())
    
    time.sleep(1)

```

This is the result of the code execution:
```
Endpoints other than open requested but private key and certificate are not set.
Listening on 10.15.11.175:55102
Server started at opc.tcp://10.15.11.175:55102
0
0
0
etc.
```

There are only zeros instead of real values.
How the given sensor tag (B2 DB11.DBX3.0	X1010.3) is introduced/linked in FreeOPCUA server? Any help very much appreciated.

petric3, 2019-09-13:
@oroulet It supports Ethernet communication protocol.

oroulet, 2019-09-13:
in your example over you do not set any value so it is 0 by default. you need to call Temp1.set_value(something). when do get this *something* value we do n ot know it is depending on your PLC. if it has ethernet, it must have som protocl over it. you need to find out, this is not related to opcua

AndreasHeine, 2019-11-11:
1. For cyclic updating i reccomend you to use the "Thread VarUpdater" like in the server-example
2. You never write to the node! 
Example for a Int:
value=33
Temp1.set_value(ua.DataValue(ua.Variant(value, ua.VariantType.Int)))
In combination with the "Thread VarUpdater" it would solve your updating problem!

Maybe this helps!
https://github.com/AndreasHeine/SecurePythonOpcUaServer
https://github.com/AndreasHeine/PythonOpcUaServer-for-Simatic-S7-1500-OpcUaClient
https://github.com/AndreasHeine/SecurePythonOpcUaClient

zerox1212, 2019-11-12:
There is a simple example (that is out of date) you could look at there https://github.com/zerox1212/python-opcua-modbus-driver