FreeOpcUa/python-opcua #891: how to use call_method that without arguments

maomaopenjun, 2019-09-17:
hello, guys
I am using UAexpert to make a simulation. and I have written a python programm to read the fake data.  this is OK
second, i need to call a method to stop the fake machine
that is the problem 

```
mycondit= myvar.get_child("3:AirConditioner_1")
mycondit.call_method("3: AirConditioner_1.Stop", *args)
```
[image]

as i know,there is no  *args to select or sth.
could you tell me how to do stop method in a right form?

oroulet, 2019-09-17:
Try with nothing

maomaopenjun, 2019-09-17:
it did not work...

maomaopenjun, 2019-09-18:
shall i set the args with VariantType like Int64, Int 32？

oroulet, 2019-09-18:
if your method does not want any arg then the correct syntax is `        job_id = self.node.call_method(f"{self.ua_idx}:Stop")`. this is taken from a working project
if it takes args, you can set the type if needed

maomaopenjun, 2019-09-18:
thanks for reply

[image]
syntax is :
```
mynode = client.get_node("3:AirConditioner_1")
something_stop = mynode.call_method(f"{self.ua_idx}:Stop")
```
what should i write in {**self.ua_idx**}？ 
sorry, I am a new bird

tdesroches, 2019-09-30:
@maomaopenjun 

I have some functions that don't require args, I used this to call it.
[image]
```
# Get Parent Node of Method
sys = client.get_node("ns=16;s=DiagnosisLogbook")

# Get Method Node
method = client.get_node("ns=16;s=DiagnosisLogbook.ClearError")

#Call Method
sys.call_method(method)
```

maomaopenjun, 2019-09-30:
thx  iwill try

closed 2019-10-21 by maomaopenjun