FreeOpcUa/python-opcua #895: Calling Method with an Arg

tdesroches, 2019-09-27:
I'm trying to pass an argument to a method.  I was able to call another method that didn't require an argument but I'm not sure how to add the arg.  If someone could give me some pointers I'd appreciate it.

[image]
[image]
[image]

This is what I'm trying.  uaexpert can call the method with the argument as expected. 
```
sys = client.get_node("ns=18;s=System")
erclr = await sys.get_child("ns=18;s=System.SetTargetMode")
args = ua.Variant([4,])
await sys.call_method(erclr, args)
```
The Response
[image]

tdesroches, 2019-09-30:
When I call the method through uaexpert I get this 
[image]

It shows the datatype as "SystemModeEnum", this is a custom datatype.  I think I might need to convert the value to this datatype before passing to the call_method().  After load_type_definitions() I'm not sure how to specify the custom type...

Type Node
[image]

EnumValues 
[image]

tdesroches, 2019-10-16:
I made a mistake in the code above.  The code I have now is
[image]
The result
[image]

I think it may have to do with the variant type...seen above uaExpert calls the data type SystemModeEnum.

tdesroches, 2019-10-16:
I used the opcua-client-gui, from there I see the arg type is specifed as ns=17;i=3014.  How do I specify a custom datatype?

tdesroches, 2019-10-17:
Ok Solved.  For those maybe struggling as I have.

Calling a method is easy, especially if it uses native Variant types, however in my case the method I was attempting to call used a custom Variant that was defined on the server.  In order to use this custom Variant I had to get the type node from the server convert it to a variant then use it to define the datatype of my arg when calling the method.

Below is my example, this is a bosch motion controller calling the method to change the controller mode.

```
from opcua import Client, ua
from opcua.common.ua_utils import data_type_to_variant_type

if __name__ == "__main__":
    url = 'opc.tcp://192.168.60.1:4840'
    client = Client(url)
    try:
        client.connect()

        custype = client.get_node("ns=17;i=3014")
        sys = client.get_node("ns=18;s=System")
        Mode = client.get_node("ns=18;s=System.SetTargetMode")
        sys.call_method(Mode, ua.Variant(4, data_type_to_variant_type(custype)))

    finally:
        client.disconnect()
```

closed 2019-10-17 by tdesroches

zerox1212, 2019-10-17:
Custom types are annoying. It's a pain to handle systems with many custom types when in most cases the native ones would be just fine.

tdesroches, 2019-10-17:
Yea it was very frustrating...at the end of the day I sent an integer...
I'm glad I got it, at least if someone struggles with it in the future they can look here.

hbszjjw, 2023-06-02:
I have a similar problem that I have a custome dataype ScanData which is a Union type as follows:
```
ScanData: Union
  ByteString: ByteString
  String: String
  Epc: ScanDataEpc
  Custom: BaseDataType
```
actually I just need set its value to null in UaExpert, but I don't know how to do it in Python.

schroeder-, 2023-06-02:
ua.ScanData() = Null

schroeder-, 2023-06-02:
Sry this only works in Asyncua

hbszjjw, 2023-06-02:
Sorry, I don't understand, do you mean I should import Asyncua instead of opcua package?