FreeOpcUa/python-opcua #898: Adding other Server as Child

Robinosius, 2019-09-30:
Is there a way of adding a different server as a child of a node in another servers namespace?
Is there some kind of "best practice" for this? Layering servers is part of OPC UA Specification (Part 1, P.16), and is really practical to combine the functionality of several devices.

oroulet, 2019-09-30:
never tried or thought about it. would be interresting to try and see what is missing

Robinosius, 2019-09-30:
There are approaches of aggregating servers by other researchers. But these are not well documented. In a paper I found on the topic, the address space of the child servers was scanned and then recreated in the top server, linking the created objects through clients. It would be interesting to see if there are more straight forward solutions to this challenge

ondrejnovakcvut, 2019-11-13:
Duplicate of #705