FreeOpcUa/python-opcua #909: Method execution not working on clients "BadNothingToDo"

sagilar, 2019-11-08:
Hi,
I am trying to run some methods which were declared on my test opc ua network (basing on the examples) but they do not work.
I call the methods on the clients but I always get "`BadNothingToDo: There was nothing to do because the client passed a list of operations with no elements.(BadNothingToDo)`". I saw the solution posted [here](https://github.com/FreeOpcUa/python-opcua/issues/297), but it seems not to be applicable for my requirements, because I need to call the methods on the clients.
Of course, I pass the input parameters when calling the methods.

Both alternatives present the same mistake:  
```
client_object_method = client_object.add_method(idx,"ClientSideMethod",client_method,[ua.VariantType.Int64],[ua.VariantType.Int64])

client_object_method = client_object.add_method(idx,"ClientSideMethod",client_method,[inargx],[outarg])
```
and the same occurs with the functions:
```
@uamethod  
def client_method(parent,u):  
    return u+1

def client_method(parent,variant):  
    return ua.Variant(variant.Value+1,ua.VariantType.Int64)  
```

Then, when I try to call the method on the object, it throws the error:  
```
client_object = devices_folder_node.add_object(idx,"NodeCreatedByClient")

client_object.call_method("{}:ClientSideMethod".format(idx),5)
```

I should mention that everything on the network is running right; the problem appears just when calling the services on the network (for every available service the problem appears).  

I hope you can help me with this issue.  
Many thanks in advance.

oroulet, 2019-11-08:
That error ALWAYS means that you not linked the method node to a python function on server side. Test examples first then try to spot the differences

sagilar, 2019-11-08:
Ok. When the methods are declared on the server code it's right. I used the `server.link_method(node,function)` and it worked. But I have a method declared on a client, and the function is declared only there. The server does not know anything about the function (which intends to exploit SOA capabilities). How could I access the server from the client to call the `link_method` on the client side? Or must I declare the function on the server side too for the method to work successfully?  
 I am a little confused if this were the only way for the methods to execute on a distributed SO network. It would be strange to declare the method twice.  
  
Thanks for your help again.

oroulet, 2019-11-08:
Then I am confused. A method is ALWAYS defined in server side. Clients can only called method defined on server. This is the basis of client/server architecture. Even a fundamental property of object oriented programming...and SOA

sagilar, 2019-11-08:
Then, the only way that a client can spread a service on the network is making it client and server at the same time? It would be nice if clients can also offer services for the other clients on the network; the processing distribution and distributed AI would be highly improved in the systems.  
So, there is no way to call the `link_method` on the client side? or another similar method?
I was thinking something like `client.get_server().link_method(...)`.  
Do you think it is possible?  
  
Thanks again.

oroulet, 2019-11-08:
Then you need to have both client client and server on some process. This is completely fine. Most complicated software are both

sagilar, 2019-11-08:
Is there a way to get an instance of the server in a specific client? Or is it possible to create another server but connected to the same network of the original server (2 servers on the same network)?  
  
Thanks for your help. It was very useful.

oroulet, 2019-11-08:
As long as two servers have different ports on same host

sagilar, 2019-11-08:
I will remain unsatisfied because I won't be able to deploy services on the client side on the same network :(.  
I will go on looking for some strategy to do it.

oroulet, 2019-11-08:
Yes you can. Just take another port

sagilar, 2019-11-08:
Ok I'll try. Another option that I see is to complement this gap implementing REST services on server side functions which call the real functions of the clients, but that's not the idea.

zerox1212, 2019-11-12:
If all you want is to trigger a method that lives on the Client, you could still do it in a non standard way.  
  
For example, your client could subscribe to a "trigger node", and when that value goes to True you could run a local method, then write a response back to the server with the result.  
  
Otherwise @oroulet is correct. If your client needs proper UA methods, then it must also run a server. This isn't even difficult to do because `client` and `server` class are already together with many common objects.

AndreasHeine, 2019-11-12:
Use a strict SOA! 
"OPC UA Webhookstyle" sounds like bad structure... and of course headaches.
Implement client/server and server/client!

keep it simple 😉