FreeOpcUa/python-opcua #914: How to set the OPC UA Server information?

pangshuai, 2019-11-23:
Hi!

I am very appreciated that your work has helped me a lot! But I have encourtered a problem that happens when the client tried to build a connection with the server-example. I can’t see any OPC UA Server information in the client as shown in the figure below. 
[image]
And I saw this function has been implemented in Server.py.  But I don't know why it didn't work. Can you help me with this question? I'll be grateful!
[image]

AndreasHeine, 2019-11-23:
Hey,

this is how i set for example stuff in the server object:

Server:
    servicelevel_node = server.get_node("ns=0;i=2267") #Service-Level Node
    servicelevel_value = 255 #0-255 [Byte]
    servicelevel_dv = ua.DataValue(ua.Variant(servicelevel_value, ua.VariantType.Byte))
    servicelevel_node.set_value(servicelevel_dv)
Client:
    servicelevel_node = client.get_node("ns=0;i=2267")
    servicelevel_node_value = servicelevel_node.get_value()

hope its helpful!

pangshuai, 2019-11-23:
First of all, thank you very much!
I will have a try. Best wish to you!

> Hey,
>
> this is how i set for example stuff in the server object:
>
> Server:
> servicelevel_node = server.get_node("ns=0;i=2267") #Service-Level Node
> servicelevel_value = 255 #0-255 [Byte]
> servicelevel_dv [...]