FreeOpcUa/python-opcua #916: Documentation: "How to generate certificate"

AndreasHeine, 2019-11-28:
Hey,

i came across this manual for creating a x509 cert-file:
https://github.com/FreeOpcUa/python-opcua/blob/master/examples/generate_certificate.sh

first of all it will work, but not with all other opc ua stacks out there!
you can test it with UAExpert it marks the missing subAltname/UA Extension (it will work if you ignore it) but... (-_-)°

if you use a x509v3 you will have no trouble with other stacks (HowTo below).

---

**Generate your own Certificate x509v3**
Step 1: Change ssl.conf (subjectAltname, country, organizationName, ...)
Step 2: openssl genrsa -out key.pem 2048
Step 3: openssl req -x509 -days 365 -new -out certificate.pem -key key.pem -config ssl.conf

**ssl.conf:**
[ req ]
default_bits = 2048
default_md = sha256
distinguished_name = subject
req_extensions = req_ext
x509_extensions = req_ext
string_mask = utf8only
prompt = no

[ req_ext ]
basicConstraints = CA:FALSE
nsCertType = client, server
keyUsage = nonRepudiation, digitalSignature, keyEncipherment, dataEncipherment, keyCertSign
extendedKeyUsage= serverAuth, clientAuth
nsComment = "OpenSSL Generated Certificat"
subjectKeyIdentifier=hash
authorityKeyIdentifier=keyid,issuer
subjectAltName = URI:urn:opcua:python:server,IP: 127.0.0.1

[ subject ]
countryName = DE
stateOrProvinceName = HE
localityName = HE
organizationName = AndreasHeine
commonName = PythonOpcUaServer

closed 2020-06-01 by AndreasHeine

ksrangini05, 2020-11-18:
Hi,,I am getting an error "BadCertificateHostNameInvalid"  in UA expert tool while connecting opc server with self signed certificate. In configuration file also, I have added  :  subjectAltName = URI:urn:opcua:python:server,IP: 127.0.0.1, "My company Name" in ssl config. Please someone help on this

AndreasHeine, 2020-11-18:
python opcua server?

ksrangini05, 2020-11-18:
yes, python opcua server

AndreasHeine, 2020-11-18:
is the cert for auth or encryption?

ksrangini05, 2020-11-18:
- for encryption
- if I want to use the same self signed certificate (from OPENSSL) to connect local host server & cloud server(basically have different IP addresses), how can I add in the config file?

AndreasHeine, 2020-11-18:
please loock up openssl docs and the x509v3 standard! i will try to reproduce the error but most of the time is a wrong configuration in the cert causing issues

AndreasHeine, 2020-11-18:
[image]
[image]
[image]

the code i used no issues... must be your config!
https://github.com/AndreasHeine/SecurePythonOpcUaServer

ksrangini05, 2020-11-18:
ok,,,thank you

admolina19, 2023-03-27:
https://github.com/AndreasHeine/SecurePythonOpcUaServer

El repositorio esta caido talvez alguien podria ayudarme estoy intenado replicar pero aun no e tneido exito y una pregunta se puede generar el certificado de esta manera

https://vscode.dev/github/admolina19/CodeOPCUA/blob/7b78f5347656a4fc587415da25972e2aadf0a860/Curso/Certificate.py#L1-L51

schroeder-, 2023-03-28:
san = "URI:urn:opcua:testserver, IP:0.0.0.0"
As of spec san must begin with uri of the server.
Also you din't set the server urn to match san.