FreeOpcUa/python-opcua #917: B&R PLC Connecting Issue

elitezhu, 2019-11-29:
Hi, freeOPCUA team,
I'm trying to connect to B&R PLC via freeOPCUA. But in the process of connecting, it fails. But UAexpert succeeds to connect it. The log is as below:
[2019-11-29 18:44:12,173 | ua_client.py | 140]: opening connection
[2019-11-29 18:44:12,190 | ua_client.py | 93]: Thread started
[2019-11-29 18:44:12,190 | connection.py | 304]: Waiting for header
[2019-11-29 18:44:12,317 | connection.py | 306]: received header: Header(type:b'ACK', chunk_type:b'F', body_size:20, channel:0)
[2019-11-29 18:44:12,318 | connection.py | 304]: Waiting for header
[2019-11-29 18:44:12,318 | ua_client.py | 181]: open_secure_channel
[2019-11-29 18:44:12,319 | ua_client.py | 53]: Sending: OpenSecureChannelRequest(TypeId:FourByteNodeId(i=446), RequestHeader:RequestHeader(AuthenticationToken:TwoByteNodeId(i=0), Timestamp:2019-11-29 10:44:12.319588, RequestHandle:1, ReturnDiagnostics:0, AuditEntryId:None, TimeoutHint:1000, AdditionalHeader:ExtensionObject(TypeId:TwoByteNodeId(i=0), Encoding:0, None bytes)), Parameters:OpenSecureChannelParameters(ClientProtocolVersion:0, RequestType:SecurityTokenRequestType.Issue, SecurityMode:MessageSecurityMode.None_, ClientNonce:b'', RequestedLifetime:3600000))
[2019-11-29 18:44:12,323 | connection.py | 306]: received header: Header(type:b'OPN', chunk_type:b'F', body_size:123, channel:16777266)
[2019-11-29 18:44:12,324 | connection.py | 304]: Waiting for header
[2019-11-29 18:44:12,325 | ua_client.py | 264]: create_session
[2019-11-29 18:44:12,325 | ua_client.py | 53]: Sending: CreateSessionRequest(TypeId:FourByteNodeId(i=461), RequestHeader:RequestHeader(AuthenticationToken:TwoByteNodeId(i=0), Timestamp:2019-11-29 10:44:12.325582, RequestHandle:2, ReturnDiagnostics:0, AuditEntryId:None, TimeoutHint:1000, AdditionalHeader:ExtensionObject(TypeId:TwoByteNodeId(i=0), Encoding:0, None bytes)), Parameters:CreateSessionParameters(ClientDescription:ApplicationDescription(ApplicationUri:urn:freeopcua:client, ProductUri:urn:freeopcua.github.io:client, ApplicationName:LocalizedText(Encoding:2, Locale:None, Text:Pure Python Client), ApplicationType:ApplicationType.Client, GatewayServerUri:None, DiscoveryProfileUri:None, DiscoveryUrls:[]), ServerUri:None, EndpointUrl:opc.tcp://192.168.31.180:4840, SessionName:Pure Python Client Session1, ClientNonce:b'~\xe0\x8bl\xcf\x99\x1d\x07$\xc8Z\xd40\xeb\x86\xb4\x03\x93T\x81\x1b7(\x8d\x11\xbe\x1a=\x10}\xf4`', ClientCertificate:None, RequestedSessionTimeout:3600000, MaxResponseMessageSize:0))
[2019-11-29 18:44:12,332 | connection.py | 306]: received header: Header(type:b'ERR', chunk_type:b'F', body_size:8, channel:0)
[2019-11-29 18:44:12,339 | connection.py | 293]: Received an error: MessageAbort(error:StatusCode(Good), reason:)
[2019-11-29 18:44:12,340 | ua_client.py | 113]: Received an error: MessageAbort(error:StatusCode(Good), reason:)
[2019-11-29 18:44:12,341 | ua_client.py | 101]: Protocol Error
Traceback (most recent call last):
  File "E:\PyEnv\python3.7\lib\site-packages\opcua\client\ua_client.py", line 96, in _run
    self._receive()
  File "E:\PyEnv\python3.7\lib\site-packages\opcua\client\ua_client.py", line 114, in _receive
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "E:\PyEnv\python3.7\lib\site-packages\opcua\client\ua_client.py", line 124, in _call_callback
    .format(request_id, self._callbackmap.keys())
opcua.ua.uaerrors._base.UaError: No future object found for request: 0, callbacks in list are dict_keys([2])
[2019-11-29 18:44:12,343 | connection.py | 304]: Waiting for header
[2019-11-29 18:44:12,484 | ua_client.py | 98]: Socket has closed connection
[2019-11-29 18:44:12,485 | ua_client.py | 102]: Thread ended
[2019-11-29 18:44:16,327 | ua_client.py | 151]: Request to close socket received
[2019-11-29 18:44:16,327 | ua_client.py | 161]: Socket closed, waiting for receiver thread to terminate...
[2019-11-29 18:44:16,328 | ua_client.py | 165]: Done closing socket: Receiving thread terminated, socket disconnected
[2019-11-29 18:44:16,329 | opcua_op.py | 124]: Connect OPCUA server opc.tcp://192.168.31.180:4840... FAIL.

And I compare the telegram of both, the difference has 2 points:
1. the clientnonce field set 32 bytes 0 in uaexpert.
2. I use the None policy, freeOPCUA fills the clientcertificate with 4 bytes 0. But uaexpert still fill the field with its own client certificate.

Could you give some hints?

oroulet, 2019-12-01:
Look at code in UaClient and change the values to something your server wants. Then make a PR and we can have a look at it.

elitezhu, 2019-12-02:
@oroulet 
I changed the two fields similar to uaexpert. But it still fails to create session.
here is the telegram by Wireshark.

Please help to have a check.

[source telegram.zip](https://github.com/FreeOpcUa/python-opcua/files/3909136/source.telegram.zip)

oroulet, 2019-12-02:
If it does not work, then the issue is somewhere else . Can you send successful Wireshark log with uaexpert and a failed one with python. I can try to have a look

MaoniinoaM, 2019-12-02:
@oroulet ，
Thank you. The log is as below:

[FreeOPCUA_Uaexpert_log.zip](https://github.com/FreeOpcUa/python-opcua/files/3909771/FreeOPCUA_Uaexpert_log.zip)