FreeOpcUa/python-opcua #918: connect to siemens s7 300

gharsallahmoez, 2019-11-30:
how can I get data from siemens s7 300 through python-opcua

AndreasHeine, 2019-11-30:
i know for the s7-400 is a cp 434-1 opc ua model avalible but i cant tell you that the cp 343-1 is also opc ua avalible... the cheapest way is to use a raspberry pi (or a similar device Maybe a more rugged one like the RevisionPi Connect) with snap7 to access the plc variables and serve them with python opc ua server!

gharsallahmoez, 2019-12-01:
I'll try it if you give me the client and server python source code of s7-400, because I think I can run it also with the s7-300. thanks

gharsallahmoez, 2019-12-01:
thanks @AndreasHeine for your reply ,
I'll try it if you give me the client and server python source code of s7-400, because I think I can run it also with the s7-300.

AndreasHeine, 2019-12-01:
if you plan to use a Simatic CP (OPC UA Version) you need to configure it as server or client depending on what you want to do! if i just want just data from the plc, i would configure the CP as a OPC UA Server (look in the siemens manual for the CP). if its up and running you can use UAExpert for example to access the OPC UA Server grab the namespace index and the node ids and put it in the sample client of python-opcua!