FreeOpcUa/python-opcua #919: client connect kapware, set_value return  BadWriteNotSupported

xyx518809, 2019-12-03:
I try to write a data to kapware, use the function set_value.
if I write like this:

```python
var4 = client.get_node('ns=2;s=13#Package01.13#Package01.D11900')
datavalue = ua.DataValue(ua.Variant(1, ua.VariantType.Int16))
datavalue.SourceTimestamp = datetime.utcnow()
var4.set_value(datavalue)
```

the program will raise an error.

```shell
opcua.ua.uaerrors._auto.BadWriteNotSupported: The server does not support writing the combination of value, status and timestamps provided.(BadWriteNotSupported)
```

and if I write like this:

```python
var4 = client.get_node('ns=2;s=13#Package01.13#Package01.D11900')
datavalue = ua.DataValue(ua.Variant(1, ua.VariantType.Int16))
#datavalue.SourceTimestamp = datetime.utcnow()
var4.set_value(datavalue)
```

The program is OK.

I want to know why this happened, thank you.

zerox1212, 2019-12-03:
Probably because Kepware OPC UA is junk. What happens if you write value UaExpert? Does it send SoureTimestamp that works?

oroulet, 2019-12-03:
this has nothing to do with our software. It is limitation of kepware. Send them emails so maybe one day they will fix their server

schroeder-, 2019-12-03:
It's not kepwares fault. From OPC UA Specification Part 4:

>  If the SourceTimestamp or the ServerTimestamp is specified, the Server shall use these values. The Server returns a Bad_WriteNotSupported error if it does not support writing of timestamps.

oroulet, 2019-12-03:
Exactly what i wrote

zerox1212, 2019-12-03:
@schroeder- What "production" server wouldn't even support timestamps lol?

schroeder-, 2019-12-03:
You don't want the server timestamp from a DataValue written by a client. Because most servers use the server timestamp for caching and subscription book keeping. Also you want the server timestamp the time when the value changed on the server, not when the client send a write request. Thats why must servers dont allow setting the server timestamp of a datavalue.

oroulet, 2019-12-03:
I think the only server I have heard complaining about our cliet behaviour is kepware.

> You don't want the server timestamp from a DataValue written by a client.

I jsut double check the code and serverTimestamp is not written by default, only the source timestamp. which, in my mind, is correct. since the progam writting is, per definiton, the source.

What do you want to change here? if you have a special server, or special requirements, you just write a DataValue and do not let python-decide guess what you want.

schroeder-, 2019-12-03:
I wouldn't set any timestamp per default as the Specification allow to reject any timestamp writes.

oroulet, 2019-12-03:
I think we tried that one. but then many servers expected it. setting it is not wrong, it is a sensible default and the API let's you do something else if you want to.
the strange/bad stuff here is that kepware does not support it. This is the only one I know of. I am sure you can find more.

zerox1212, 2019-12-03:
@schroeder- Per the standard Kepware is fine to block the write, but it's still seems lazy to me. Also, the SourceTimestamp is being written which makes sense for a Client to do. The example above is not doing anything ServerTimestamp.

schroeder-, 2019-12-03:
Yeah keep it as it is. 
You just wanted to give @xyx518809 a explanation why Kepware can disalow to set the source timestamp.