FreeOpcUa/python-opcua #920: Only username and password (withour certificate and private key)

Aburub, 2019-12-05:
Hey all,

I am trying to access SINUMERIK (Siemens) OPC UA. The server is only username and password protected (without certificate and private key). I have tried to connect with the following:

     client = Client(url)
     client.set_user("my_user")
     client.set_passwotd("my_password")
     client.connect()

but I have received a "TimeoutError"
I have also tried the following:

     client = Client(url)
     client.connect_socket()
     client.create_session()
     client.activate_session(username="my_user", password = "my_password")

and I also received TimeoutError
can someone please help me ?
PS: I am new to this field

Thanks

zerox1212, 2019-12-05:
Test the server with UaExpert first. However if you are getting timeout it's probably a networking issue.

Aburub, 2019-12-05:
It is working perfectly on UAExpert.

aigefjk, 2019-12-18:
The user identity token is valid but the server has rejected it.(BadIdentityTokenRejected)

aigefjk, 2019-12-18:
> The user identity token is valid but the server has rejected it.(BadIdentityTokenRejected)

this is  the problem which I met using：

 client = Client(url)
 client.set_user("my_user")
 client.set_passwotd("my_password")
 client.connect()

AndreasHeine, 2019-12-18:
client.set_user("user1")
client.set_password("pw1")
**client.set_security_string("Basic256Sha256,Sign,cert.der,key.pem")**
client.connect()

AndreasHeine, 2019-12-18:
BadIdentity could be a certificate issue!

Interstella12, 2020-03-30:
Same problem. Did you solve it?

lightbreakerD, 2021-04-09:
I have the same issue when trying to connect to a S7-1200.

AndreasHeine, 2021-04-09:
it would be important to post the full stacktrace to tell us the pyrhon-opcua version and how the opc ua server is configured (messagesecuritymode, ...) we only know that you want to user user/pw auth ... if you use sign an encrypt: do you have created your own cert and key which matches the ip and applicationuri of your client pc? its not possible to help you with that little information... otherwise its just quessing 😅