FreeOpcUa/python-opcua #921: Exception client.Disconnect() if subscriptions are used

EEflow, 2019-12-06:
Disconnection a simple client with subscriptions results in a error since version 0.98.8.
I don't think this is a big issues but want to report it anyway. Maybe the error the error is linked to more serious problem.

```python
import time
import opcua

url = "opc.tcp://localhost:4840/"

class MySubHandler(object):

    def datachange_notification(self, node, val, data):
        print("New data change event", node, val)

client = opcua.Client(url)
client.connect()

server_node = client.get_server_node()
node = server_node.get_child(["ServerStatus", "CurrentTime"])

my_subscribtion = client.create_subscription(100, MySubHandler())
my_subscribtion.subscribe_data_change(node)

time.sleep(3)

print("Disconnect..")
client.disconnect()

```

Results in the following output:
```
New data change event Node(NumericNodeId(i=2258)) 2019-12-06 13:16:08.963409
New data change event Node(NumericNodeId(i=2258)) 2019-12-06 13:16:09.979448
New data change event Node(NumericNodeId(i=2258)) 2019-12-06 13:16:10.981524
New data change event Node(NumericNodeId(i=2258)) 2019-12-06 13:16:11.987235
Disconnect..
exception calling callback for <Future at 0x1723e29c548 state=cancelled>
Traceback (most recent call last):
  File "C:\home\user\anaconda3\envs\opcua\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\home\user\anaconda3\envs\opcua\lib\site-packages\opcua\client\ua_client.py", line 476, in _call_publish_callback
    data = future.result()
  File "C:\home\user\anaconda3\envs\opcua\lib\concurrent\futures\_base.py", line 426, in result
    raise CancelledError()
concurrent.futures._base.CancelledError
exception calling callback for <Future at 0x1723e29c708 state=cancelled>
Traceback (most recent call last):
  File "C:\home\user\anaconda3\envs\opcua\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\home\user\anaconda3\envs\opcua\lib\site-packages\opcua\client\ua_client.py", line 476, in _call_publish_callback
    data = future.result()
  File "C:\home\user\anaconda3\envs\opcua\lib\concurrent\futures\_base.py", line 426, in result
    raise CancelledError()
concurrent.futures._base.CancelledError
```

AndreasHeine, 2019-12-18:
try unsubscribe before disconnecting !?

EEflow, 2019-12-18:
Same result if using unsubscribe:

```python
import time
import opcua

url = "opc.tcp://localhost:4840/"

class MySubHandler(object):

    def datachange_notification(self, node, val, data):
        print("New data change event", node, val)

client = opcua.Client(url)
client.connect()

server_node = client.get_server_node()
node = server_node.get_child(["ServerStatus", "CurrentTime"])

my_subscribtion = client.create_subscription(100, MySubHandler())
handler = my_subscribtion.subscribe_data_change(node)

time.sleep(2)

print("Unsubscribe..")
my_subscribtion.unsubscribe(handler)

time.sleep(2)

print("Disconnect..")
client.disconnect()
```

```
New data change event Node(NumericNodeId(i=2258)) 2019-12-18 08:40:31.435586
New data change event Node(NumericNodeId(i=2258)) 2019-12-18 08:40:32.429028
New data change event Node(NumericNodeId(i=2258)) 2019-12-18 08:40:33.438392
New data change event Node(NumericNodeId(i=2258)) 2019-12-18 08:40:34.440670
Unsubscribe..
exception calling callback for <Future at 0x1957422b6c8 state=cancelled>
Traceback (most recent call last):
  File "C:\home\user\anaconda3\envs\opcua\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\home\user\anaconda3\envs\opcua\lib\site-packages\opcua\client\ua_client.py", line 476, in _call_publish_callback
    data = future.result()
  File "C:\home\user\anaconda3\envs\opcua\lib\concurrent\futures\_base.py", line 426, in result
    raise CancelledError()
concurrent.futures._base.CancelledError
exception calling callback for <Future at 0x1957422b288 state=cancelled>
Traceback (most recent call last):
  File "C:\home\user\anaconda3\envs\opcua\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\home\user\anaconda3\envs\opcua\lib\site-packages\opcua\client\ua_client.py", line 476, in _call_publish_callback
    data = future.result()
  File "C:\home\user\anaconda3\envs\opcua\lib\concurrent\futures\_base.py", line 426, in result
    raise CancelledError()
concurrent.futures._base.CancelledError
Disconnect..
```

oroulet, 2019-12-18:
I just saw that one... must be one the recent pull requests...

AndreasHeine, 2019-12-18:
New data change event Node(NumericNodeId(i=2258)) 2019-12-18 08:50:34.987547
New data change event Node(NumericNodeId(i=2258)) 2019-12-18 08:50:35.961039
New data change event Node(NumericNodeId(i=2258)) 2019-12-18 08:50:36.974321
Unsubscribe..
Disconnect..
PS C:\Users\HeineAn>

AndreasHeine, 2019-12-18:
no issues if i use your code!

oroulet, 2019-12-18:
@AndreasHeine I get that error on master

oroulet, 2019-12-18:
@brubbel I suppose this is due to https://github.com/FreeOpcUa/python-opcua/commit/837377f44b97f9b73d2f1b5ae72e02e5d6776988

I am not really sure what the correct behavior is. We cancel a task so we get a cancellError this is normal

oroulet, 2019-12-18:
OK catchintg it: https://github.com/FreeOpcUa/python-opcua/pull/924

AndreasHeine, 2019-12-18:
@oroulet 
0.98.7 => No Error
0.98.8 => Same Error

oroulet, 2019-12-18:
and 0.98.9 -> No Error now :tada:

closed 2019-12-18 by oroulet

AndreasHeine, 2019-12-18:
👍

pinkynrg, 2020-04-22:
I still get this error once in a while using version 0.98.9:

```
[JOB_EXECUTER  ] Received an error: MessageAbort(error:StatusCode(BadRequestTypeInvalid), reason:None)
[JOB_EXECUTER  ] Received an error: MessageAbort(error:StatusCode(BadRequestTypeInvalid), reason:None)
[JOB_EXECUTER  ] Protocol Error
[JOB_EXECUTER  ] Traceback (most recent call last):
[JOB_EXECUTER  ]   File "/usr/local/lib/python3.7/dist-packages/opcua/client/ua_client.py", line 101, in _run
[JOB_EXECUTER  ]     self._receive()
[JOB_EXECUTER  ]   File "/usr/local/lib/python3.7/dist-packages/opcua/client/ua_client.py", line 121, in _receive
[JOB_EXECUTER  ]     self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
[JOB_EXECUTER  ]   File "/usr/local/lib/python3.7/dist-packages/opcua/client/ua_client.py", line 131, in _call_callback
[JOB_EXECUTER  ]     .format(request_id, self._callbackmap.keys())
[JOB_EXECUTER  ] opcua.ua.uaerrors._base.UaError: No future object found for request: 0, callbacks in list are dict_keys([2, 3])
[JOB_EXECUTER  ] Exception in thread Thread-1499:
[JOB_EXECUTER  ] Traceback (most recent call last):
[JOB_EXECUTER  ]   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
[JOB_EXECUTER  ]     self.run()
[JOB_EXECUTER  ]   File "/usr/local/lib/python3.7/dist-packages/opcua/client/client.py", line 64, in run
[JOB_EXECUTER  ]     self.client.open_secure_channel(renew=True)
[JOB_EXECUTER  ]   File "/usr/local/lib/python3.7/dist-packages/opcua/client/client.py", line 309, in open_secure_channel
[JOB_EXECUTER  ]     result = self.uaclient.open_secure_channel(params)
[JOB_EXECUTER  ]   File "/usr/local/lib/python3.7/dist-packages/opcua/client/ua_client.py", line 265, in open_secure_channel
[JOB_EXECUTER  ]     return self._uasocket.open_secure_channel(params)
[JOB_EXECUTER  ]   File "/usr/local/lib/python3.7/dist-packages/opcua/client/ua_client.py", line 199, in open_secure_channel
[JOB_EXECUTER  ]     response = struct_from_binary(ua.OpenSecureChannelResponse, future.result(self.timeout))
[JOB_EXECUTER  ]   File "/usr/lib/python3.7/concurrent/futures/_base.py", line 430, in result
[JOB_EXECUTER  ]     raise CancelledError()
[JOB_EXECUTER  ] concurrent.futures._base.CancelledError
[JOB_EXECUTER  ]
```