FreeOpcUa/python-opcua #925: How to determine the variable size of subscription

JHQRAIN, 2019-12-27:
I want use a value val ,I want determine the variable size of subscription# '' print("Python: New data change event", node, val)''

import sys
sys.path.insert(0, "..")
import logging

from opcua import Client

class SubHandler(object):

    """
    Client to subscription. It will receive events from server
    """

    def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)

    def event_notification(self, event):
        print("Python: New event", event)

if __name__ == "__main__":
    #from IPython import embed
    logging.basicConfig(level=logging.WARN)
    client = Client("opc.tcp://192.168.56.100:49320/OPCUA/SimulationServer/")
    #client = Client("opc.tcp://192.168.56.100:4840/OPCUA/SimulationServer/")
    #client = Client("opc.tcp://olivier:olivierpass@localhost:53530/OPCUA/SimulationServer/")
    try:
        client.connect()
        root = client.get_root_node()
        print("Root is", root)
        print("childs of root are: ", root.get_children())
        print("name of root is", root.get_browse_name())
        objects = client.get_objects_node()
        print("childs og objects are: ", objects.get_children())

        tag1 = client.get_node("ns=2;s=Channel1.Device1.Tag1")
        print("tag1 is: {0} with value {1} ".format(tag1, tag1.get_value()))
        tag2 = client.get_node("ns=2;s=Channel1.Device1.Tag2")
        print("tag2 is: {0} with value {1} ".format(tag2, tag2.get_value()))

        handler = SubHandler()
        sub = client.create_subscription(500, handler)
        handle1 = sub.subscribe_data_change(tag1)
        handle2 = sub.subscribe_data_change(tag2)

        from IPython import embed
        embed()

        
        sub.unsubscribe(handle1)
        sub.unsubscribe(handle2)
        sub.delete()
    finally:
        client.disconnect()

zerox1212, 2019-12-27:
I have no idea what you are asking for and your code isn't formatted at all.

JHQRAIN, 2019-12-30:
Thank you for answering my question
I'm sorry！I really want to ask：
from opcua import Client

class SubHandler(object):
    def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)        
    def event_notification(self, event):
        print("Python: New event", event)
    

client = Client("opc.tcp://127.0.0.1:49320")
client.connect()
tag1 = client.get_node("ns=2;s=test.test.VW102")
handler = SubHandler()
sub = client.create_subscription(200, handler)
handle1 = sub.subscribe_data_change(tag1)

I pass “print("Python: New data change event", node, val)”，
I can get it“Python: New data change event Node(StringNodeId(ns=2;s=test.test.VW102)) 592”，butI think in the main function：“a =val， print("Python: New data change event", a)”。
I can pass “a” ，get the value of node change

JHQRAIN, 2019-12-30:
Thank you for answering my question！Well, I mean：“how to print(v) in the main function  ”:

from opcua import Client
class SubHandler(object):
    global v   
    def datachange_notification(self, node, val, data):
        global v
        v = val
        #print("Python: New data change event", node, v)
        return v
client = Client("opc.tcp://127.0.0.1:49320")
client.connect()
tag2 = client.get_node("ns=2;s=test.test.VW100")
handler = SubHandler()
sub = client.create_subscription(200, handler)
handle1 = sub.subscribe_data_change(tag2)
print(v)

JHQRAIN, 2019-12-30:
Thank you for answering my question！Well, I mean：“how to print(v) in the main function ”:
from opcua import Client

class SubHandler(object):
    def datachange_notification(self, node, val, data):
        global v
        v = val
        #print("Python: New data change event", node, v)
        return v

client = Client("opc.tcp://127.0.0.1:49320")
client.connect()
tag2 = client.get_node("ns=2;s=test.test.VW100")
handler = SubHandler()
sub = client.create_subscription(200, handler)
handle1 = sub.subscribe_data_change(tag2)
print(v)