FreeOpcUa/python-opcua #942: Can't run opcua-client.exe example

borissheikman, 2020-01-16:
I want to learn how to establish a connection between my machine and a remote OPC UA server. The opcua-client.exe example looked like a good place to start experimenting and learning. Unfortunately, I am having troubles running the program. I installed it via pip :

pip install freeopcua-client

I get the following error when I run it. Sorry for any lines that wrap around. :)

PS C:\Users\ME\AppData\Local\Programs\Python\Python38\Scripts> .\opcua-client.exe
Traceback (most recent call last):
  File "C:\Users\ME\AppData\Local\Programs\Python\Python38\Scripts\opcua-client-script.py", line 11, in <module>
    load_entry_point('opcua-client==0.8.0', 'console_scripts', 'opcua-client')()
  File "c:\users\ME\appdata\local\programs\python\python38\lib\site-packages\pkg_resources\__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "c:\users\ME\appdata\local\programs\python\python38\lib\site-packages\pkg_resources\__init__.py", line 2852, in load_entry_point
    return ep.load()
  File "c:\users\ME\appdata\local\programs\python\python38\lib\site-packages\pkg_resources\__init__.py", line 2443, in load
    return self.resolve()
  File "c:\users\ME\appdata\local\programs\python\python38\lib\site-packages\pkg_resources\__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "c:\users\ME\appdata\local\programs\python\python38\lib\site-packages\uaclient\mainwindow.py", line 20, in <module>
    from uaclient.graphwidget import GraphUI
  File "c:\users\ME\appdata\local\programs\python\python38\lib\site-packages\uaclient\graphwidget.py", line 16, in <module>
    import pyqtgraph as pg
  File "c:\users\ME\appdata\local\programs\python\python38\lib\site-packages\pyqtgraph\__init__.py", line 204, in <module>
    from .graphicsItems.VTickGroup import *
  File "c:\users\ME\appdata\local\programs\python\python38\lib\site-packages\pyqtgraph\graphicsItems\VTickGroup.py", line 7, in <module>
    from .. import functions as fn
  File "c:\users\ME\appdata\local\programs\python\python38\lib\site-packages\pyqtgraph\functions.py", line 17, in <module>
    from . import debug
  File "c:\users\ME\appdata\local\programs\python\python38\lib\site-packages\pyqtgraph\debug.py", line 11, in <module>
    from . import ptime
  File "c:\users\ME\appdata\local\programs\python\python38\lib\site-packages\pyqtgraph\ptime.py", line 24, in <module>
    cstart = systime.clock()  ### Required to start the clock in windows
AttributeError: module 'time' has no attribute 'clock'

I suppose the last line is the most import. 

Am I missing a library somewhere? 

BTW, I have experience with the OPC DA library and was able to establish local connections within one machine. This is a bit of a leap for me, going from DA to UA and going from one machine to two machines.

Thanks!

-=- Boris

zerox1212, 2020-01-17:
Seems the `time.clock()` function was removed in Python 3.8. Run the application with Python 3.7 or lower and it should work. This looks to be a Python 3.8 compatibility issue with PYQT.

borissheikman, 2020-01-17:
You're right! I bumped down to 3.7 and the client seems to at least partially work. The graphing doesn't work but I think it's because PyQt5 is installed for Python 3.8. 

Thanks!