FreeOpcUa/python-opcua #943: SendBufferSize vs ReceiveBufferSize

brubbel, 2020-01-19:
https://github.com/FreeOpcUa/python-opcua/blob/ec188d15072f39602ba1f977a2ea97fd4de8b513/opcua/common/connection.py#L340-L343

When the client receives an ACK message from the server, it stores the ACK.SendBufferSize as _max_chunk_size (used to chunk messages which it sends to the receiver.)
However, SendBufferSize is the maximum message size the server will send to the client.
https://reference.opcfoundation.org/v104/Core/docs/Part6/7.1.2/#7.1.2.4

So, instead this should be `self._max_chunk_size = msg.ReceiveBufferSize`

oroulet, 2020-01-19:
Sounds like a big error ;-) thanks

oroulet, 2020-01-19:
It looks like the entire chunk code must be check. We also sometimes get strange errors from that code complaining about wrong sequence

brubbel, 2020-01-20:
Reading carefully, it seems that the current code is correct. The client sends a Hello + SendBufferSize1 to the server, which replies with an ACK + SendBufferSize2 <= SendBufferSize1. Then the client sets this SendBufferSize2 as the maximum TX size.
(If I understand correctly the SendBufferSize and ReceiveBufferSize are symmetrical for client and server, but they first negotiate the maximum values)

closed 2020-02-23 by brubbel