FreeOpcUa/python-opcua #944: Not able to continuous run the opc client

juhiajmera13, 2020-01-22:
Hi

I am trying to run the client code for a long time, but it is shutting down every now and then. Can someone help?

oroulet, 2020-01-22:
you need to write down your setup, what server, what authentication, encryption
and of course send the stack trace here

juhiajmera13, 2020-01-22:
This is what I am trying to do:

var=client.get_node("ns=4;i=4000")
client=Client(url)
client.connect()
print("Connected")
try:
    while True:
          print(var.get_value())
except ConnectionError as e:    # This is the correct syntax
   print(e)
   r = "No response"

I want to run this for a few days to get data from the machine. But, the code is getting shut down every now and then randomly. The errors what I am getting are:

File "<ipython-input-3-c604fa2d7fd3>", line 1, in <module>
    runfile('C:/Users/E0559351/Documents/AMMachine/client_v2.python.py', wdir='C:/Users/E0559351/Documents/AMMachine')

  File "C:\Users\E0559351\AppData\Local\Continuum\anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 827, in runfile
    execfile(filename, namespace)

  File "C:\Users\E0559351\AppData\Local\Continuum\anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/E0559351/Documents/AMMachine/client_v2.python.py", line 19, in <module>
    client.connect()

  File "C:\Users\E0559351\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\client\client.py", line 257, in connect
    self.send_hello()

  File "C:\Users\E0559351\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\client\client.py", line 289, in send_hello
    ack = self.uaclient.send_hello(self.server_url.geturl(), self.max_messagesize, self.max_chunkcount)

  File "C:\Users\E0559351\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\client\ua_client.py", line 251, in send_hello
    return self._uasocket.send_hello(url, max_messagesize, max_chunkcount)

  File "C:\Users\E0559351\AppData\Local\Continuum\anaconda3\lib\site-packages\opcua\client\ua_client.py", line 177, in send_hello
    ack = future.result(self.timeout)

  File "C:\Users\E0559351\AppData\Local\Continuum\anaconda3\lib\concurrent\futures\_base.py", line 437, in result
    raise TimeoutError()

TimeoutError

juhiajmera13, 2020-01-22:
I want to run the client application continuously but the client disconnects and doesn't automatically connect.

jehanjoshi007, 2020-03-04:
Is this issue resolved ?

oroulet, 2020-03-04:
This big report does not point to an issue. It is too vague. You can get timeout for 1000s reasons

fcaspani01, 2020-03-04:
Hi @oroulet,
I have the same issue of @jehanjoshi007.
Can I increase timeout value for get_value() function?

oroulet, 2020-03-04:
No. It does not help anything. The timeout just says the communication is down. Increasing the timeout just warns you later. There is probably another stack trace before telling you what the real error is. Maybe some buffer or socket error or server error

fcaspani01, 2020-03-04:
Ok thanks, what do you suggest in this case?

oroulet, 2020-03-04:
Wait for next crash and find the real stop reason. Then open a bug report

oroulet, 2020-03-04:
Must be another error BEFORE the timeout error

oroulet, 2020-03-04:
Yes. You can also serve the Wireshark trace here. This is the best debug data we can get

jehanjoshi007, 2020-03-04:
@oroulet  thanks the next time I face the same issue I will share the wireshark trace here.

jehanjoshi007, 2020-03-24:
@oroulet I was able to reproduce the issue this was the error I got the code apparently runs for a few weeks and then raises a timeout error 

`import pyodbc
import psycopg2 
import pandas as pd 
import datetime as dt
import collections 
import opcua
from opcua import *
import socket
import sys 
import urllib.parse
import time, uuid
import urllib.request
import hmac, hashlib
from base64 import b64encode
from json import loads
import concurrent.futures
import gc
import numpy as np
clientc = Client("")
clientc.connect()
def database_connection():

    try:
    
        conn = psycopg2.connect("")
        cur = conn.cursor()
    
        return cur,conn

    except:

        dc = "error"

    

def plant_temp():
    try:
        ts = str(int(time.time()))
        with urllib.request.urlopen() as response:
            html = response.read()
            data= html.decode('utf-8')

   
        l = []
        l.append(data)
#     seq1 ='bTemperature'
#     seq2 ='\nc'
        l = [item.replace(' ','') for item in l]
        l = [item.split('\n') for item in l]

    #print(l)
        for ele in l:
        #print(ele)
    
            Temperature = ele[1][12:16]
            Humidity = ele[2][9:13]
            Dewpoint = ele[3][9:13]
            response.close()
    
        return Temperature,Humidity,Dewpoint

    except:
        print("Error")
    

def get_temp_hum():

    try:
[... 514 lines cut ...]
            main()
            #gc.collect()
            time.sleep(55)
        except:
            m = "error"`

File "C:\Users\jehan.SKAPS\Desktop\PythonOPC\OPC_LiveV10.py", line 39, in <module>
    clientc.connect()

  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\client\client.py", line 257, in connect
    self.send_hello()

  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\client\client.py", line 289, in send_hello
    ack = self.uaclient.send_hello(self.server_url.geturl(), self.max_messagesize, self.max_chunkcount)

  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\client\ua_client.py", line 251, in send_hello
    return self._uasocket.send_hello(url, max_messagesize, max_chunkcount)

  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\client\ua_client.py", line 177, in send_hello
    ack = future.result(self.timeout)

  File "C:\ProgramData\Anaconda3\lib\concurrent\futures\_base.py", line 434, in result
    raise TimeoutError()

TimeoutError

maintainenceskaps, 2020-06-01:
@oroulet  when we run the script the fetch data continuously we get the following error as well and we cannot connect to the opc ua server 

>>> from opcua import *
>>> clientc = Client("opc.tcp://10.1.4.208:4870")
>>> clientc.connect()

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\client\client.py", line 274, in connect
    self.send_hello()
  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\client\client.py", line 306, in send_hello
    ack = self.uaclient.send_hello(self.server_url.geturl(), self.max_messagesize, self.max_chunkcount)
  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\client\ua_client.py", line 262, in send_hello
    return self._uasocket.send_hello(url, max_messagesize, max_chunkcount)
  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\client\ua_client.py", line 190, in send_hello
    ack = future.result(self.timeout)
  File "C:\ProgramData\Anaconda3\lib\concurrent\futures\_base.py", line 430, in result
    raise CancelledError()
concurrent.futures._base.CancelledError

Interesting thing is the dashboard is built in node red and node red has no issues at all in maintaining a continuous connection with the opc ua server so this should not be a server issue

oroulet, 2020-06-02:
From the stack above you are not running for a few weeks but trying to connect and failing. If you are not running continuously then just add a try and except. And retry any errors.
But I have no ideas why the server refuses the connection. This is before any session creation it sounds more like a network error or similar

maintainenceskaps, 2020-06-02:
@oroulet It was running continuously and then the timeout error happens I have try and except but it keeps trying and I keep seeing a timeout error this is only resolved when I restart the opc ua server which is in a Siemens HMI screen. The python exe file I have created runs for a few days and then timeout error occurs after which until I restart the server it will never connect back 

[TimeOutSelected1.zip](https://github.com/FreeOpcUa/python-opcua/files/4717345/TimeOutSelected1.zip)

I ran this stack to just show the stack trace I have attached the wireshark trace as well if that would be helpful to figure out what happened.