FreeOpcUa/python-opcua #946: Add unit / extensionObject / Engineering Unit

qualityland27, 2020-01-26:
Hello to everybody,

I am trying to add a unit (engineering unit like millimeter or degree celsius) to variables. 
For this I think I have to set the DataType of the
variable to an extensionobject with value "EUInformation". And this 
"EUInformation" can be filled with the correct unit.
I was searching in the web for some examples but no success. I also tried to 
add_property and figure it out by try and error but also no success. 

Can anybody here help me out?

Here is some of the code I tried. I tried to add thu unit "pascal" to the variable "Pressure".

```
painting = node.add_folder(addspace, "Painting")
sprayer = painting.add_object(addspace, "Sprayer")
Pres = sprayer.add_variable(addspace, "Pressure", 0)

obj = ua.ExtensionObject()
        obj.Body = 
b'EUInformation{NamespaceUri=http://www.opcfoundation.org/UA/units/un/cefact, UnitId=5259596, DisplayName=LocalizedText{text=Pa, locale=null}, Description=LocalizedText{text=pascal, locale=null}}'
        obj.TypeId = ua.NodeId.from_string('ns=0;i=887')
        obj.Encoding = 1
        obj._freeze = True

unit1 = obj
pressProperty = Pres.add_property(addspace, "EUInformation", ua.Variant([unit1])) 
```

Thank a lot in advance!

oroulet, 2020-01-26:
Is unit something standardized in UA? If not just use a string :. pressProperty = Pres.add_property(addspace, "EUInformation", "Euro")
Or use an enum (i think there is an example for enums in example folder)

qualityland27, 2020-01-26:
Thanks for fast answering. 

I tried the idea with enum but its not what I want to do. Neighter is the idea with using a String what I am looking for. 

As I understood, a way to add units to a variable is using the extensionObject as datatype of the variable.

oroulet, 2020-01-27:
Of you really want to use a structure (encoded as extension object ) you need to create the structure using some is modeller )or by hand

oroulet, 2020-01-27:
But if you simply want some custom datatypes then there is probably an example for that. But playing with extension objects directly is probably not what you want

oroulet, 2020-01-27:
Ok maybe I understand. You want to create a new datatype. Then create your variable using that datatypes. Look at example for custom datatypes