FreeOpcUa/python-opcua #955: I am trying to run opcua simple server in linux and getting some error.

jadhavshital, 2020-02-06:
Hi,
I have installed two packages to run opcua server.
1) `sudo pip install opcua`
[image]

2) `sudo pip install cryptography`
[image]

After this I am trying to run my opcua server but getting below error
[image]

My source code :- 
[opcua_server_encryption.txt](https://github.com/FreeOpcUa/python-opcua/files/4164687/opcua_server_encryption.txt)

Is it issue cryptography package version? Currently I have installed the latest one(2.8)
And the same code works in windows but in windows I have installed cryptography with command `pip install cryptography==2.4` and this same command does not work in linux.

Can anyone tell me what can be the solution to run server script on linux?

Thanks in advanced.

SebastianDD, 2020-02-07:
Hi,
this shouldn't be an issue of the cryptography package. I run a docker container with python:3.7-slim-buster image and did install cryptography>=2.7.

It seems that you should check if all other libs are available --> 'lxml>=4.4.0'

oroulet, 2020-02-07:
pip3 install lxml should work, that is what everybody is using here. But I heard that maybe there was some issue on windows...

zerox1212, 2020-02-07:
@oroulet pip3 install lxml on windows will compile some C++ /.net type libraries. I don't remember exactly, but there are extra dependencies for lxml on windows.

brubbel, 2020-02-08:
In my experience, lxml installed via apt does not work on all architectures.
What does work:
python3 -m pip install lxml

jadhavshital, 2020-02-10:
Previously my lxml version was 3.5.0, now I have installed 4.4.3 and opcua server script is running now.
Thanks for the help.

closed 2020-02-10 by jadhavshital