FreeOpcUa/python-opcua #956: OPC UA Server: AttributeError: 'NoneType' object has no attribute 'encrypted_block_size'

AndreasHeine, 2020-02-08:
Has anyone encountered this error?

It happens after a long while, i listen/subscribed to a variable witch randomly changes over a long period of time...

It looks like the subscription break because of a nonetype -> he trys to encrypt a nonetype object "return self.Encryptor.encrypted_block_size()"

Server:
Python 3.7.2 64bit
Python OPC UA 0.98.9
Sign&Encrypt + UserAuth

Client:
Node-Red IIoT opcua

both on localhost!

Server Error:
Exception in callback InternalSubscription._sub_loop()
handle: <TimerHandle when=71435.953 InternalSubscription._sub_loop()>
Traceback (most recent call last):
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\asyncio\events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\server\internal_subscription.py", line 294, in _sub_loop
    self.publish_results()
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\server\internal_subscription.py", line 322, in publish_results
    self.callback(result)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\server\uaprocessor.py", line 80, in forward_publish_response
    self.send_response(requestdata.requesthdr.RequestHandle, requestdata.seqhdr, response)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\server\uaprocessor.py", line 47, in send_response       
    struct_to_binary(response), message_type=msgtype, request_id=seqhdr.RequestId)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\common\connection.py", line 248, in message_to_binary   
    token_id=self.security_token.TokenId)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\common\connection.py", line 108, in message_to_chunks   
    max_size = MessageChunk.max_body_size(crypto, max_chunk_size)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\common\connection.py", line 82, in max_body_size        
    max_plain_size = (max_encrypted_size // crypto.encrypted_block_size()) * crypto.plain_block_size()
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\crypto\security_policies.py", line 125, in encrypted_block_size
    return self.Encryptor.encrypted_block_size()
AttributeError: 'NoneType' object has no attribute 'encrypted_block_size'

Server-Code:
https://github.com/AndreasHeine/opcua-mes/blob/master/app.py

Node-Red Flow:
[{"id":"bece226e.58ffc","type":"OPCUA-IIoT-Listener","z":"d688b3ff.efcb4","connector":"288b4a9d.5bbcee","action":"subscribe","queueSize":10,"name":"","topic":"","justValue":true,"useGroupItems":false,"showStatusActivities":false,"showErrors":false,"x":280,"y":360,"wires":[["82fb54f8.e7f77"]]},{"id":"33309510.b0c3fa","type":"ui_chart","z":"d688b3ff.efcb4","name":"","group":"55a0b09c.0ce3f8","order":0,"width":0,"height":0,"label":"chart","chartType":"line","legend":"false","xformat":"HH:mm:ss","interpolate":"linear","nodata":"","dot":false,"ymin":"0","ymax":"150","removeOlder":1,"removeOlderPoints":"","removeOlderUnit":"3600","cutout":0,"useOneColor":false,"colors":["#1f77b4","#aec7e8","#ff7f0e","#2ca02c","#98df8a","#d62728","#ff9896","#9467bd","#c5b0d5"],"useOldStyle":false,"outputs":1,"x":830,"y":360,"wires":[[]]},{"id":"9158bf6.b7bb84","type":"function","z":"d688b3ff.efcb4","name":"get_listener_value","func":"msg.payload = msg.payload.value[1];\nreturn msg;","outputs":1,"noerr":0,"x":650,"y":360,"wires":[["33309510.b0c3fa"]]},{"id":"3f0486b8.e562b2","type":"OPCUA-IIoT-Inject","z":"d688b3ff.efcb4","injectType":"listen","payloadType":"num","topic":"","repeat":"","crontab":"","once":true,"startDelay":"10","name":"","payload":"1000","addressSpaceItems":[{"name":"Random","nodeId":"ns=2;i=6","datatypeName":""}],"x":130,"y":360,"wires":[["bece226e.58ffc"]]},{"id":"82fb54f8.e7f77","type":"OPCUA-IIoT-Response","z":"d688b3ff.efcb4","name":"","compressStructure":true,"showStatusActivities":false,"showErrors":false,"activateUnsetFilter":false,"activateFilters":false,"negateFilter":false,"filters":[],"x":450,"y":360,"wires":[["9158bf6.b7bb84"]]},{"id":"288b4a9d.5bbcee","type":"OPCUA-IIoT-Connector","z":"","discoveryUrl":"","endpoint":"opc.tcp://localhost:4840","keepSessionAlive":true,"loginEnabled":true,"securityPolicy":"Basic256Sha256","securityMode":"SIGNANDENCRYPT","name":"py","showErrors":false,"individualCerts":false,"publicCertificateFile":"","privateKeyFile":"","defaultSecureTokenLifetime":"","endpointMustExist":false,"autoSelectRightEndpoint":false,"strategyMaxRetry":"","strategyInitialDelay":"","strategyMaxDelay":"","strategyRandomisationFactor":"","requestedSessionTimeout":"","connectionStartDelay":"","reconnectDelay":"","maxBadSessionRequests":"10"},{"id":"55a0b09c.0ce3f8","type":"ui_group","name":"Group 1","tab":"d14b2992.730a9","order":1,"disp":true,"width":6},{"id":"d14b2992.730a9","type":"ui_tab","name":"Tab 1","icon":"dashboard","order":1}]

brubbel, 2020-02-08:
You could test without security enabled. The renewal of secure channels is broken at the moment. It will cause the connection to break whenever the client renews the channel. For UaExpert, this is after 45mins.
https://github.com/FreeOpcUa/python-opcua/issues/926

AndreasHeine, 2020-02-08:
ok!
does it also explain this:

Traceback (most recent call last):
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\server\binary_server_asyncio.py", line 75, in _process_data   
    ret = self.processor.process(hdr, buf)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\server\uaprocessor.py", line 83, in process
    msg = self._connection.receive_from_header_and_body(header, body)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\common\connection.py", line 335, in receive_from_header_and_body
    return self._receive(chunk)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\common\connection.py", line 366, in _receive
    self._check_incoming_chunk(msg)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\common\connection.py", line 311, in _check_incoming_chunk     
    .format(self._peer_sequence_number, num))
opcua.ua.uaerrors._base.UaError: Wrong sequence 111 -> 170 (server bug or replay attack)

oroulet, 2020-02-08:
I think the last one is something else. Do you have any idea when you get it?

AndreasHeine, 2020-02-08:
hard to say comes and goes ... not reliable reproduceable ... now it runs since hours without any problem!
never mind, thanks!

oroulet, 2020-02-24:
@brubbel is renewale of security broken? I tried to test everyhtin gI could the last days with servers and clients and did not manage to get any issue, using very short timeouts and renewing all the times. What do I have to do to reproduce?

brubbel, 2020-03-02:
1. uaexpert (settings->configure): minrevisedsecurechannellifetime to e.g. 5000 milliseconds.
2. freeopcua (connection.py): set response of open-secure-channel to 5000ms by using `self.security_token.RevisedLifetime = 5000` instead of params.RequestedLifetime by uaexpert.
3. Connect uaexpert using sign+encrypt. It will fail and restart the connection every 5 seconds instead of renewing.

closed 2020-06-01 by AndreasHeine

EEflow, 2020-07-27:
Can this one open again?

EEflow, 2020-07-27:
> @oroulet
> 
> 1. uaexpert (settings->configure): minrevisedsecurechannellifetime to e.g. 5000 milliseconds.
> 2. freeopcua (connection.py): set response of open-secure-channel to 5000ms by using `self.security_token.RevisedLifetime = 5000` instead of params.RequestedLifetime by uaexpert.
> 3. Connect uaexpert using sign+encrypt. It will fail and restart the connection every 5 seconds instead of renewing.

I can reproduce the error by changing two parameters in UaExport:

```
Stack.SecureConnection_MinRevisedSecureChannelLifetime = 5000
General.SecureChannelLifetime = 5000
```
(First change `MinRevisedSecureChannelLifetime`, then `SecureChannelLifetime `)

When the failure occurs UaExpert does not register any errors. It the values are just frozen without any sign the connection has crashed.

oroulet, 2020-07-27:
And does ut work without encryption?
Do you use master?

EEflow, 2020-07-27:
Some error messages:

**Security: None + None**

With security policy `none` everything seems fine.

**Security: Basic256sha256 + Sign**

opcua master **aed7ab0bbc8935a396b734634e378926a8dcbdf7**
```
Creating an open endpoint to the server, although encrypted endpoints are enabled.
Tried to read attribute '10' in FourByteNodeId(i=2258), but the attribute is missing
Tried to read attribute '14' in TwoByteNodeId(i=35), but the attribute is missing
Tried to read attribute '15' in TwoByteNodeId(i=35), but the attribute is missing
Tried to read attribute '16' in TwoByteNodeId(i=35), but the attribute is missing
Tried to read attribute '14' in TwoByteNodeId(i=33), but the attribute is missing
Tried to read attribute '15' in TwoByteNodeId(i=33), but the attribute is missing
Tried to read attribute '16' in TwoByteNodeId(i=33), but the attribute is missing
Tried to read attribute '14' in TwoByteNodeId(i=31), but the attribute is missing
Tried to read attribute '15' in TwoByteNodeId(i=31), but the attribute is missing
Tried to read attribute '16' in TwoByteNodeId(i=31), but the attribute is missing
Tried to read attribute '14' in TwoByteNodeId(i=58), but the attribute is missing
Tried to read attribute '15' in TwoByteNodeId(i=58), but the attribute is missing
Tried to read attribute '16' in TwoByteNodeId(i=58), but the attribute is missing
Tried to read attribute '10' in TwoByteNodeId(i=58), but the attribute is missing
Tried to read attribute '14' in FourByteNodeId(i=2004), but the attribute is missing
Tried to read attribute '15' in FourByteNodeId(i=2004), but the attribute is missing
Tried to read attribute '16' in FourByteNodeId(i=2004), but the attribute is missing
Tried to read attribute '10' in FourByteNodeId(i=2004), but the attribute is missing
Exception in callback InternalSubscription._sub_loop()
handle: <TimerHandle when=381454.619446846 InternalSubscription._sub_loop()>
Traceback (most recent call last):
  File "/home/user/miniconda3/lib/python3.7/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/home/user/Documents/python/opcua_test/dependencies/python-opcua/opcua/server/internal_subscription.py", line 293, in _sub_loop
    self.publish_results()
  File "/home/user/Documents/python/opcua_test/dependencies/python-opcua/opcua/server/internal_subscription.py", line 321, in publish_results
    self.callback(result)
  File "/home/user/Documents/python/opcua_test/dependencies/python-opcua/opcua/server/uaprocessor.py", line 80, in forward_publish_response
    self.send_response(requestdata.requesthdr.RequestHandle, requestdata.seqhdr, response)
  File "/home/user/Documents/python/opcua_test/dependencies/python-opcua/opcua/server/uaprocessor.py", line 47, in send_response
    struct_to_binary(response), message_type=msgtype, request_id=seqhdr.RequestId)
  File "/home/user/Documents/python/opcua_test/dependencies/python-opcua/opcua/common/connection.py", line 237, in message_to_binary
    chunks = MessageChunk.message_to_chunks(self.security_policy, message, self._max_chunk_size, message_type=message_type, channel_id=self.security_token.ChannelId, request_id=request_id, token_id=self.security_token.TokenId)
  File "/home/user/Documents/python/opcua_test/dependencies/python-opcua/opcua/common/connection.py", line 105, in message_to_chunks
    max_size = MessageChunk.max_body_size(crypto, max_chunk_size)
  File "/home/user/Documents/python/opcua_test/dependencies/python-opcua/opcua/common/connection.py", line 81, in max_body_size
    return max_plain_size - ua.SequenceHeader.max_size() - crypto.signature_size() - crypto.min_padding_size()
  File "/home/user/Documents/python/opcua_test/dependencies/python-opcua/opcua/crypto/security_policies.py", line 153, in signature_size
    return self.Signer.signature_size()
AttributeError: 'NoneType' object has no attribute 'signature_size'
```

**Security: Basic256sha256 + Sign & Encrypt**

Tested with 2 version (master and last release):

opcua master **aed7ab0bbc8935a396b734634e378926a8dcbdf7**
```
Tried to read attribute '14' in FourByteNodeId(i=2004), but the attribute is missing
Tried to read attribute '15' in FourByteNodeId(i=2004), but the attribute is missing
Tried to read attribute '16' in FourByteNodeId(i=2004), but the attribute is missing
Tried to read attribute '10' in FourByteNodeId(i=2004), but the attribute is missing
Tried to read attribute '14' in TwoByteNodeId(i=58), but the attribute is missing
Tried to read attribute '15' in TwoByteNodeId(i=58), but the attribute is missing
Tried to read attribute '16' in TwoByteNodeId(i=58), but the attribute is missing
Tried to read attribute '10' in TwoByteNodeId(i=58), but the attribute is missing
Exception in callback InternalSubscription._sub_loop()
handle: <TimerHandle when=380808.40091346897 InternalSubscription._sub_loop()>
Traceback (most recent call last):
[... 21 lines cut ...]
opcua tag **0.98.12**
``` 
Exception in callback InternalSubscription._sub_loop()
handle: <TimerHandle when=380455.701013272 InternalSubscription._sub_loop()>
Traceback (most recent call last):
  File "/home/user/miniconda3/lib/python3.7/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/home/user/opcua_test/dependencies/python-opcua/opcua/server/internal_subscription.py", line 293, in _sub_loop
    self.publish_results()
[... cut ...]
    max_size = MessageChunk.max_body_size(crypto, max_chunk_size)
  File "/home/user/opcua_test/dependencies/python-opcua/opcua/common/connection.py", line 80, in max_body_size
    max_plain_size = (max_encrypted_size // crypto.encrypted_block_size()) * crypto.plain_block_size()
  File "/home/user/opcua_test/dependencies/python-opcua/opcua/crypto/security_policies.py", line 126, in encrypted_block_size
    return self.Encryptor.encrypted_block_size()
AttributeError: 'NoneType' object has no attribute 'encrypted_block_size'
```

EEflow, 2020-07-27:
Sorry, I meant to post this in https://github.com/FreeOpcUa/python-opcua/issues/926 (which is actually open..)