FreeOpcUa/python-opcua #957: Logging OPCUA time series data and logging timer issues

smjure, 2020-02-09:
Perhaps this question is not directly related to this module but may be of interest in the future. Also, here probably are Python's OPCUA users from  industry to share some experiences and help.

I trying to switch from Matlab and its OPC toolbox to Python's OPCUA. I successfully setup your OPCUA client and started to log time series data (just values) via numpy.savetxt(). Matlab has logger that stores time series data via timer function into .olf file. Besides node values, it contains nodeIDs, time stamp, quality and EventTime, so several times more data. Comparison showed that my files containing the same parameters with same sampling rate and in double precision are of about the same size as Matlab's .olf (~300MB/day of data). Obviously, numpy.savetxt() is not an optimal function for data logging not to mention I skipped other attributes. I have done a lot of search but not found any good solution. I wonder how you log your time series data from OPC servers?

Second thing, I have doubts about the logger's timer I applied. I searched a lot and eventually apply similar solution to [this](https://stackoverflow.com/a/25251804/6213592)

```
dt = 1.0 # sampling time
starttime = time.time()
while True:   # logging loop
    code for reading OPC data     
    numpy.savetxt() # logging data

    time.sleep(dt - ((time.time() - starttime) % dt)) # timer function, 'trying' to reduce calculation delay

```
By printing out the time I saw that it has up to 0.12s inaccuracy (which is too much, I have 1s sampling rate). Does the OPCUA has any option for timer or sampling time specification or how do you solve this issues in your problem? 

Thank you for sharing your experiences in advance, very much appreciated.

oroulet, 2020-02-09:
Comming from matlab and intresting in data analysys you should use pandaslibrary instead of numpy.
https://pandas.pydata.org/  Also look at jupyter notebook. Matlab people like it usually.

Also I have no idea about what matlab saves but numpy saved exactly what you ask it to save. nothing more and nothing else.If you want to save space save as binary and not txt. Although if you save data all the time I would rather save data in a sqlite database instead of a simple text file. and make backups.

Also concerning opcua, if you want to save all values for a given node you should use subscriptions at a given frequency. not a loop. Look at a subscription example. every node has a timestamp, what precision you get i depening on how your application is made. It is not realated to opcua or this particular opcua library.

oroulet, 2020-02-09:
also logging the time you get the data in your example it probably not what you want to do. you want to use one of the timestamps

zerox1212, 2020-02-09:
For the record, python isn't very good with precision timing in general. Other than your timing solution you would probably need to split your program into two parts. One that saves the values in memory, and another one that happens somewhere else (like a thread) that writes the output.