FreeOpcUa/python-opcua #958: Receiving subscription notification although value did not change

ap-0, 2020-02-11:
This issue is related to the version in use: 0.98.7

**1. Usage description.**

I am using this OPC-UA library in a productive environment where the production
on multiple PLCs is monitored. One OPC-UA server provides nodes of a various
number of operation points. Each operation point has a numeric status node
which changes depending on its working state (1 idle, 10 working, 3 and 4
malfunction, etc.)

There are 4 OPC-UA servers on different PLCs with different IPs.

I have a total of 8 Python processes running. Source code of all processes is identical.
Each process creats a subscription (configured in a JSON file and passed as
argument to the process) on a various number of status nodes of the operation
points.

Proc. 1 - Connects to PLC1 - 32 Subscriptions
Proc. 2 - Connects to PLC1 -  1 Subscriptions
Proc. 3 - Connects to PLC1 -  7 Subscriptions
Proc. 4 - Connects to PLC2 - 26 Subscriptions
Proc. 5 - Connects to PLC3 - 61 Subscriptions
Proc. 6 - Connects to PLC3 -  1 Subscriptions
Proc. 7 - Connects to PLC3 -  5 Subscriptions
Proc. 8 - Connects to PLC4 - 22 Subscriptions

Each data change notification is processed in a thread that is created by me.
This thread reads some additional information from the PLC (multiple values in
nodes on the OPC-UA server) which then creates an entry in a python Queue. This
queue is processed by a single dedicated thread which writes data into the
database.

**All 8 processes are started (or stopped) through a shell script one after
another in a very short interval.
This seems to be an important detail for my issue.**

**2. Unpredicted behavior**

Besides the Python processes I have written this simple script which creates a
single subscription to a status node on for example PLC1. Each time the status
changes it displays the new status with a time stamp.

```
    class TestSubHandler():
        def datachange_notification(self, node, value, data):
            print(datetime.now(), "Data change", value)

    client = Client("opc.tcp://0.0.0.0:4840")
    client.connect()

    node_path = ["Path", "To", "Node"]
    sub_node = client.get_root_node().get_child(node_path)
    sub = client.create_subscription(500, TestSubHandler())
    handle = sub.subscribe_data_change(sub_node)

    try:
        while True:
            time.sleep(10)
    except KeyboardInterrupt:
        client.disconnect()
```

If working correctly the output looks usually like this (exemplary):

```
    2020-02-11 15:11:27.ms Data change 1
    2020-02-11 15:11:27.ms Data change 10
    2020-02-11 15:11:27.ms Data change 1
    2020-02-11 15:11:28.ms Data change 10
    2020-02-11 15:11:28.ms Data change 1
    2020-02-11 15:11:29.ms Data change 10
    2020-02-11 15:11:29.ms Data change 1
    2020-02-11 15:11:30.ms Data change 10
    2020-02-11 15:11:30.ms Data change 1
    2020-02-11 15:11:31.ms Data change 10
    2020-02-11 15:11:31.ms Data change 1
```

While this script is running I start all my 8 Python processes through the
script and suddenly the output becomes

```
    2020-02-11 15:11:27.ms Data change 1
    2020-02-11 15:11:27.ms Data change 10
    2020-02-11 15:11:28.ms Data change 1
    2020-02-11 15:11:28.ms Data change 10  # Processes are started after this one!
    2020-02-11 15:11:28.ms Data change 1
    2020-02-11 15:11:29.ms Data change 1
    2020-02-11 15:11:29.ms Data change 1
    2020-02-11 15:11:30.ms Data change 1
    2020-02-11 15:11:30.ms Data change 10
    2020-02-11 15:11:30.ms Data change 10
    2020-02-11 15:11:31.ms Data change 10
    2020-02-11 15:11:31.ms Data change 10
    2020-02-11 15:11:31.ms Data change 10
    2020-02-11 15:11:32.ms Data change 10
    2020-02-11 15:11:33.ms Data change 1
    2020-02-11 15:11:33.ms Data change 1
    2020-02-11 15:11:33.ms Data change 1
    2020-02-11 15:11:34.ms Data change 10
    2020-02-11 15:11:34.ms Data change 10
    2020-02-11 15:11:34.ms Data change 10
    2020-02-11 15:11:35.ms Data change 10
    2020-02-11 15:11:35.ms Data change 10
    2020-02-11 15:11:35.ms Data change 1
    2020-02-11 15:11:36.ms Data change 1
    2020-02-11 15:11:36.ms Data change 1
    2020-02-11 15:11:36.ms Data change 10
    2020-02-11 15:11:37.ms Data change 10
    2020-02-11 15:11:37.ms Data change 1
```

So for some reason although the value of the node has not changed the server is
sending me a notification that it did.

**Even stranger is the fact that the false notificatons stop again after I stop
my Python processes.**

```
    2020-02-11 15:11:27.ms Data change 1
    2020-02-11 15:11:27.ms Data change 10
    2020-02-11 15:11:28.ms Data change 1
    2020-02-11 15:11:28.ms Data change 10  # Processes are started after this one!
    2020-02-11 15:11:28.ms Data change 1
    2020-02-11 15:11:29.ms Data change 1
    2020-02-11 15:11:29.ms Data change 1
    2020-02-11 15:11:30.ms Data change 1
    2020-02-11 15:11:30.ms Data change 10
    2020-02-11 15:11:30.ms Data change 10
    2020-02-11 15:11:31.ms Data change 10
    2020-02-11 15:11:31.ms Data change 10
    2020-02-11 15:11:31.ms Data change 10
    2020-02-11 15:11:32.ms Data change 10
    2020-02-11 15:11:33.ms Data change 1
    2020-02-11 15:11:33.ms Data change 1
    2020-02-11 15:11:33.ms Data change 1
    2020-02-11 15:11:34.ms Data change 10
    2020-02-11 15:11:34.ms Data change 10
    2020-02-11 15:11:34.ms Data change 10
    2020-02-11 15:11:35.ms Data change 10  # Processes are stopped here!
    2020-02-11 15:11:35.ms Data change 1
    2020-02-11 15:11:35.ms Data change 10
    2020-02-11 15:11:36.ms Data change 1
    2020-02-11 15:11:37.ms Data change 10
    2020-02-11 15:11:38.ms Data change 1
```

**If I start the processes with a sleep in between of 10 seconds (lower did not
suffice) the faulty behaviour does not occur.**

Why does the time the processes are started make a difference?
Is it a problem with the PLC? Is it a problem with the opcua library?
Did somebody ever see something like this before?

zerox1212, 2020-02-12:
Read about how python multiprocessing really works. Your processes are probably creating the entire subscription again and again.

zerox1212, 2020-02-12:
Another person tried to call UA functions from processes and also got really strange results. https://github.com/FreeOpcUa/python-opcua/issues/883

oroulet, 2020-02-12:
So you mean your simple client get extra (wrong) notifications when the server gets new connections from these 8 ua clients?

oroulet, 2020-02-12:
If you're simple clients gets these extra notifications then it sounds like an issue with your server. But it is very hard to understand your setup. You have a lot of extra not relevant information like this json config stuff

ap-0, 2020-02-12:
> 
> 
> Read about how python multiprocessing really works. If you are handing UA functions (like subscriptions) off to another process weird things can happen because multiprocessing library has an entire copy of the main processes memory. When a data change happens in a process it's function is getting called from all the processes that have a reference to it or something.

I am not using multiprocessing, only threading. And as far as I can tell freeopcua also does nowhere use multiprocessing. 

> 
> 
> So you mean your simple client get extra (wrong) notifications when the server gets new connections from these 8 ua clients?

Correct.

> 
> 
> If you're simple clients gets these extra notifications then it sounds like an issue with your server. But it is very hard to understand your setup. You have a lot of extra not relevant information like this json config stuff

The JSON is just configuration. It only contains paths to the nodes that I want the process to subscribe to. There is not much else to it.

oroulet, 2020-02-12:
So you mean your simple client get extra (wrong) notifications when the server gets new connections from these 8 ua clients?

> Correct.

Then you have a server issue. Unless your clients do something strange like wttting things

zerox1212, 2020-02-12:
Are you monitoring 32 tags with 32 individual subscriptions? Or are you using 1 subscription to monitor 32 items?

Can you check if all the subscription handles are unique? I don't have time to look in the spec, but I think every subscription gets some kind of identifier.

ap-0, 2020-02-13:
I am creation 32 individual subscriptions for 32 tags. 

I will check if I can get the identifier for each subscription and check whether there are duplicates. Thanks for the idea.

ap-0, 2020-02-17:
I've taken a look at the subscription IDs that I receive from the server and there are no duplicates. 

I have another idea but am note sure. [This](https://forum.unified-automation.com/topic545.html#p548) comment by one of the support team states that if a connection is lost it is kept open for a defined time. If a client reconnects within that time the session is reused. What about the subscriptions created in that session? Are they also reused? I often see this behaviour after a broken pipe exception after which I reconnect to the PLC and recreate subscriptions. What if I reconnect within that time frame and the subscriptions are still active but I create new subscription in addition to those that are still active. 

But this would not explain why my simple script receives more notifications without value change. It did not reconnect and recreate subscriptions.

ap-0, 2020-02-18:
Still looking for the cause and a soltuion for this problem I wanted to verify that this may indeed be a server issue. 

So I looked for another client implementation of OPC-UA. I have found [this JavaScript](https://node-opcua.github.io/) implementation. Using the [example](https://github.com/node-opcua/node-opcua/blob/v2.1.3/documentation/sample_client_ts.ts) I recreated my Python script for the JavaScript module. It does exactly the same. It connets to the same Server and subscribed to the same node for value changes. While both were running I again started all processes on the productive server. 

This is the result: 
[image]

As you can see the JavaScript OPC-UA client does not suffer from the same issue. Using UaExpert I also cannot reproduce the behavior that I get with freeopcua.

oroulet, 2020-02-18:
@ap-0 we see your screenshots, but for us to to understand you need to write some minimal test code we can run to reproduce the error.

ap-0, 2020-02-24:
@oroulet sorry that it took a while to create a minimal example. My time testing on the productive server is limited and at first I was not able to reproduce the issue with the example. And currently I am still not able to reproduce it 100 %. I am not sure why. 

With my processes running the notifications for status changes are permanent and only stop when most of the processes that all connect to the same PLC are stopped. With my minimal example I can only see the strange behavior for a couple of seconds and then it stops. Afterwards it works as intended. When starting two to three instances of my minimal example I receive 3 to 4 notifications with identital values.

Also: I am not able to reproduce this issue with the PLC I have here for testing: Siemens SIMATIC S7-1500 OPC UA, V02.01.00. While the PLC in the productive environment Siemens SIMATIC S7-1500 OPC- UA V02.05.00. 

Also also: I do get some exceptions while creating subscriptions but not on all nodes.
```python
Traceback (most recent call last):
  File "/opt/python34/lib/python3.4/concurrent/futures/_base.py", line 297, in _invoke_callbacks
    callback(self)
  File "/opt/python34/lib/python3.4/site-packages/opcua/client/ua_client.py", line 469, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "/opt/python34/lib/python3.4/site-packages/opcua/client/ua_client.py", line 88, in check_answer
    hdr.ServiceResult.check()
  File "/opt/python34/lib/python3.4/site-packages/opcua/ua/uatypes.py", line 233, in check
    raise UaStatusCodeError(self.value
```

**This is the code of the simple script of which I have posted multiple examples above:**
```python
import sys
import time
from opcua import Client
from opcua.common.subscription import Subscription
from opcua import ua
from datetime import datetime
import opcua as opc

class SubHandler():
    def datachange_notification(self, node, val, data):
        print(datetime.now(), "- Data Change - ", val)

    def event_notification(self, event):
        pass

server = "opc.tcp://<ip>:4840"
node = ["0:Objects", "3:Path", "3:To", "3:Node", "3:LK_S050", "3:Status", "3:Status"]

client = Client(server)
print("Connecting")
client.connect()
print("Done")
root = client.get_root_node()
n = root.get_child(node)
handler = SubHandler()
sub = client.create_subscription(500, handler)
handle = sub.subscribe_data_change(n)
sub.subscribe_events()

while True:
    try:
        time.sleep(10)
    except KeyboardInterrupt:
        print("CTRL+C")
        break

print("Disconnecting")
client.disconnect()
print("Done")
```

**This is the minimal example of my processes that I have broken down to the essentials as much as possible.**

```python
import sys
import json
import threading
import time
from datetime import datetime
[... 191 lines cut ...]
    for client in clients:
        client.join()
```

1. As mentioned in the beginning I get the wrong behavior only for a couple of seconds. I receive data change notifications 10, 10, 10, 10 and then it continues to work correctly.

2. I have start multiple instances of the script (2, 3 or more) for the issue to come up. If starting only one the process does not show any signs of problems. 

Below the output of the script in the first code block.
```
2020-02-24 04:02:33.966174 - Data Change -  1
2020-02-24 04:02:43.465315 - Data Change -  10
2020-02-24 04:03:39.472530 - Data Change -  1
2020-02-24 04:03:56.288786 - Data Change -  1
2020-02-24 04:03:57.298652 - Data Change -  1
2020-02-24 04:03:58.829144 - Data Change -  1
2020-02-24 04:04:05.555810 - Data Change -  1
2020-02-24 04:05:37.074840 - Data Change -  10
2020-02-24 04:06:28.585540 - Data Change -  1
2020-02-24 04:06:39.586266 - Data Change -  10
2020-02-24 04:07:46.100337 - Data Change -  1
2020-02-24 04:07:55.603414 - Data Change -  10
2020-02-24 04:08:44.113602 - Data Change -  1
2020-02-24 04:08:53.115656 - Data Change -  10
```

ap-0, 2020-02-28:
I think I can say with high confidence that is a PLC issue. As mentioned I have a testing PLC (Siemens V02.01.00) where I am not able to reproduce the issue. The produciton PLC (V02.05.00) has this issue. I have updated the testing PLC to V02.05.00 and am not able to reproduce the problem. Later today I will update to something newer than V02.05.00 and test again.

Still trange is the fact that the OPC UA client in JavaScript did not behave the same which made me think that this is not PLC related.

oroulet, 2020-02-28:
Thanks for the update. That make sense

mjk-automation-ltd, 2021-03-30:
Hi @ap-0, sorry to jump on this thread, did you determine the cause of your issue? We are facing a similar problem with a Siemens S7-1500 (6ES7 510-1DJ01-0AB0 FW V2.6) OPC-UA Server where if we subscribe to too many nodes we get the error below, which looks exactly like you described.

[image]

We think it is the normal behaviour of the OPC-UA server that it publishes all of the subscribed nodes immediately on subscription that is causing the problem, as there seems to be some sort of buffer overflow somewhere when we have too many responses to handle in a short space of time.

Did you find that a certain firmware version fixed this for you? If so, what difference in behaviour was observed, I would be interested to know if eg. FW v2.01 did not publish datachange notifications on subscription? It looks like there were some OPC changes in FW v2.5 for this CPU as there are some notes about this in the hardware catalog, so it's possible this may have been affected.

Grateful for any insight, thanks.

AndreasHeine, 2021-03-31:
maxNotificationsPerPublish!

The maximum number of notifications that the Client wishes to receive in a single Publish response. A value of zero indicates that there is no limit.
The number of notifications per Publish is the sum of monitoredItems in the DataChangeNotification and events in the EventNotificationList.

ap-0, 2021-04-03:
Hi @mjk-automation-ltd, 
no - unfortnuately the prcoesses still behave the same. I still get continues notifications although the value of the tag did not change. At first we thought we fixed the issue by updating the PLC firmware (to version 2.8 if I'm not wrong). And for a couple of days everything looked fine. The issue was not reproducable at first. But after some days it started again to send notifications.

ddodi1, 2026-01-20:
Hello,
I have a similar issue connecting to a Siemens SIMATIC S7-1500 OPC UA V04.00.00. I subscribe to 180 nodes, in my test the most part of them has stable values, while some other vary every 500 ms.
Just after the subscription opening, I get once all the 180 nodes values. This operaiton takes 4 - 5 s. Then, I receive only the notifications for the changing nodes. But, unpredictably, sometimes the handler receives a huge flow of values: all the nodes values (unchanged), sorted as in subscription opening, but arriving very quickly (one value per ms), and cyclically. After some seconds, the data flow stops and returns to notify the actually changing nodes.
Thanks for any help!

```python
class SubscriptionHandler:
    def datachange_notification(self, node: Node, value, data):
        logger.info(f'Notification: {node} {value}')
        app['queue'].put_nowait({
            'node': node,
            'value': value,
            'data': data,
        })

async def startProcess(uri, plcName):
    async with Client(url=uri) as client:
        # 180 nodes to monitor
        path = f"3:{plcName}/3:DataBlocksGlobal/3:PerfApp"
        obj = await client.nodes.objects.get_child(path)
        vars = await ua_utils.get_node_children(obj)
        
        params = ua.CreateSubscriptionParameters()
        params.RequestedPublishingInterval = 100
        params.RequestedLifetimeCount = 60
        params.RequestedMaxKeepAliveCount = 20
        params.MaxNotificationsPerPublish = 0
        params.PublishingEnabled = True
        params.Priority = 0 

        subscription = await client.create_subscription(params, SubscriptionHandler())
        for var in vars:
            await subscription.subscribe_data_change(var, sampling_interval=100)

        await asyncio.sleep(3600)
        await subscription.delete()
```