FreeOpcUa/python-opcua #959: how to reuse old session?

justin65, 2020-02-14:
When the network connection break, is it possible to reconnect server with original session?

Auto reconnect is not available now, I try to connect again manually. BUT, it seems that server side will not release old session. Sometimes I will get error 'BadTooManySessions'. Is there any implementation for reconnect with original session?

Thanks.

oroulet, 2020-02-14:
Very good question.....
Our server does not support that. It should be possible to do it from client side but I am unsure how. This will require reading spec and probably do some small changes if it does not work out of the box.
We need to recover session and subscriptions...

dns13, 2023-10-31:
Have you found a way to reuse session or correctly release the used session upon exit? Having the same issue - Server is limited to 5 concurrent sessions, after that I need to forcefully clear them to connect again.. Other UA Clients don't have that issue.

oroulet, 2023-10-31:
we are supposed to cleanly close the session if exiting the program, but you have bad network and let the session running? That is not implemented. this is open source, you are welcome to fix that, on client side it cannot be that hard
Also use opcua-asyncio it much more actively maintained. That software here is deprecated

dns13, 2023-10-31:
Oh sorry, I indeed using opcua-asyncio, wrong repo.