FreeOpcUa/python-opcua #960: Cannot read variables with [] on Sinumerik 840d

ma1ster, 2020-02-15:
Hi,

I have a bunch of variables like: 
GD5_CMM_CORR[80,1]
GD5_CMM_CORR[80,2]
GD5_CMM_CORR[80,3]
GD5_CMM_CORR[80,4]
GD5_CMM_CORR[80,5]
And what I can only see on client is one variable GD5_CMM_CORR with value 0.0.
How can I read the others ?

Sinumerik has built-in opc server

AndreasHeine, 2020-02-15:
does it work with ua expert? 
screenshot (ua expert) from the node obj you try to read?
your code?

ma1ster, 2020-02-22:
I will try it this week. Is that software free ?

oroulet, 2020-02-22:
google and you will find out quickly... yes you just need to register. or use whatever other ua client (use google too for that)

zati-, 2020-02-29:
@ma1ster did you solve your problem. We have the same problem here. Variable with bracket are non-browseable and not shown on UaExpert

oroulet, 2020-02-29:
I am guessing you do not have any variable with brackets, it is the way your server/client shows arrays. There is only one node withtout brackets

ma1ster, 2022-06-23:
@zati- sorry for late reply. When Ive added them manually they started showing values