FreeOpcUa/python-opcua #969: Cannot connect to remote OPC server: [WinError 10061]

smjure, 2020-02-24:
First of all thank you for doing such a great work.

I am trying to connect to remote OPC server. I have its Host name, Server ID, Server IP. The firewall was disconnected - all ports are opened. I tried:
```
from opcua import Client
client = Client("opc.tcp://192.168.0.50:55101") 
client.connect()
```
also
```
client = Client("opc.tcp://192.168.0.50:55101/ServerID") 
client.connect() 
```
and similar things and always get the following error:

> ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it

When I try with UaExpert, I get the following error:

> [uastack] OpcUa_SecureConnection_OnNotify: Connect event: ERROR 0x80050000!
> [uastack] OpcUa_SecureConnection_OnNotify: Connect event: Notifying owner! 0x80050000
> Could not connect to server: BadCommunicationError

On the other hand, I can easily connect with Siemens and Matlab OPC clients just by providing Host name and Server name.

Any comments or suggestions appreciated   (?)

oroulet, 2020-02-24:
Sounds like an opc da and NOT an opca ua server

smjure, 2020-02-24:
Ture, it is OPC DA server. Is there any other way to establish the connection?

brubbel, 2020-02-24:
https://github.com/FreeOpcUa/python-opcua/issues/489

smjure, 2020-02-26:
@brubbel Thank you, that worked. I'd just give a couscous reminder for future addressee to be careful about dependencies and their versions, it will save you a lot of time.

closed 2020-02-26 by smjure

LonelySadhu, 2022-07-18:
What could it be?
 raise errors.SerializeError("unsupported serialized class: " + classname)
Pyro4.errors.SerializeError: unsupported serialized class: OpenOPC.OPCError