FreeOpcUa/python-opcua #972: Bug if NodeId IdentifierType is GUID

Michael-List, 2020-02-27:
I would like to add a variable with GUID as NodeIdType:
```
my_folder.add_variable(ua.NodeId(identifier=UUID('777b9d07-9e48-4a2f-9084-7ceae7043c8a').bytes, namespaceidx=ns, nodeidtype=NodeIdType.Guid), 'guid Test', 33)
```
But I'm getting the error _AttributeError: 'bytes' object has no attribute 'time_low'_

The UUID object has the attribute time_low. I tried to create the NodeId with an UUID identifier:
```
int_identifier_folder.add_variable(ua.NodeId(identifier=UUID('777b9d07-9e48-4a2f-9084-7ceae7043c8a'), namespaceidx=ns, nodeidtype=NodeIdType.Guid), 'guid Test', 33)
```
But here im getting the error _TypeError: expected string or bytes-like object_
Which is from this file/line:
```
opcua\ua\uatypes.py", line 340, in has_null_identifier
if self.NodeIdType == NodeIdType.Guid and re.match(b'0.', self.Identifier):
```

Can someone tell me what I'm doing wrong here? Or is there a bug in the re.match?

If I change the above line from uatypes.py to this it works:
```
if self.NodeIdType == NodeIdType.Guid and re.match(b'0.', self.Identifier.bytes):
```

zerox1212, 2020-02-27:
I don't know if anyone has used UUID for NodeID. There is probably a bug here. I looked at the tests and there is a test for GUID node id, but it's being checked a different way. The test creates a `ua.GuidNodeId()` object and encode/decodes it with the binary protocol.

Michael-List, 2020-03-13:
I think I fixed it. Pull request is open

Michael-List, 2020-03-16:
Resolved in merge request #975

closed 2020-03-16 by Michael-List