FreeOpcUa/python-opcua #977: Read group or multiple variables

h1n0t0r1, 2020-03-19:
How to read group of variables or multiple variables at once ?
In OPC DA we have groups.

AndreasHeine, 2020-04-21:
Every OPC UA Variable can be an array! 
Its the most efficient way to read large amounts of data. you could combine custom data types with arrays.

AndreasHeine, 2020-04-21:
from opcua import ua, uamethod, Server

server = Server()
server.set_endpoint("opc.tcp://127.0.0.1:4840")
server.set_server_name("test")
address_space = server.register_namespace("asdf")
server.set_application_uri("urn:test")

root_node = server.get_root_node()
object_node = server.get_objects_node()
server_node = server.get_server_node()

parameter_obj = object_node.add_object(address_space, "Parameter")

array = [0,0,0,0]

test_node = parameter_obj.add_variable(address_space, "test", ua.Variant(array, ua.VariantType.UInt64))

server.start()

Brandon-lz, 2023-03-28:
you can use client.get_values([node1,node2,...])    and this will return a list of values