FreeOpcUa/python-opcua #978: Cannot connect to Kepware Server using Basic256Sha256

yanndeferne, 2020-03-21:
Hello everyone, 

I am currently implementing a secure communication (Basic256Sha256) between my opcua client (python code) and server (KEPServerEX 6). I started by confirming connection without any security policy, went on with basic 128Rsa15 and then Basic256. I am now struggling to get secure 256Sha256 communication.. What I did:
- I generated my client self-signed certificate (.der) and its private key (.pem) and imported the certificate in the trusted client list of my kepware server.
- I exported my kepware server certificate (.der) so that it can be imported in my client (python).
- The kepware server is user protected with password.

Problem: everything worked with "Basic128Rsa15" and "Basic256" protocols but I struggle to obtain communication using 256Sha256... My python code is similar to the example "client-with-encryption.py" except that I also take into consideration the server certificate. I get this error:

----------------------------
server_URL = "opc.tcp://127.0.0.1:49320"
path_server_cert = "kepware_server_certificate.der"
path_client_cert = "my_cert_example.der"
path_private_key = "private_key_example.pem"

if __name__ == "__main__":

    # Open the connection:
    client = Client(server_URL)

    client.set_security_string("Basic256Sha256,SignAndEncrypt," + path_client_cert +"," + path_private_key + "," + path_server_cert + "")

    client.application_uri = ""  #self-signed certificate generated a "" URI

    client.secure_channel_timeout = 10000
    client.session_timeout = 10000

    client.set_user("Administrator")
    client.set_password("xxxxxx")

    client.connect()

--------------------------
WARNING:opcua.client.ua_client.Socket:ServiceFault from server received  in response to ActivateSessionRequest
Traceback (most recent call last):
  File "D:/Trabajo/microred_abb/python_files/OPCUA_client_for_kepware.py", line 61, in <module>
    client.connect()
  File "D:\Trabajo\microred_abb\python_files\venv\lib\site-packages\opcua\client\client.py", line 264, in connect
    self.activate_session(username=self._username, password=self._password, certificate=self.user_certificate)
  File "D:\Trabajo\microred_abb\python_files\venv\lib\site-packages\opcua\client\client.py", line 422, in activate_session
    return self.uaclient.activate_session(params)
  File "D:\Trabajo\microred_abb\python_files\venv\lib\site-packages\opcua\client\ua_client.py", line 289, in activate_session
    data = self._uasocket.send_request(request)
  File "D:\Trabajo\microred_abb\python_files\venv\lib\site-packages\opcua\client\ua_client.py", line 84, in send_request
    self.check_answer(data, " in response to " + request.__class__.__name__)
  File "D:\Trabajo\microred_abb\python_files\venv\lib\site-packages\opcua\client\ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "D:\Trabajo\microred_abb\python_files\venv\lib\site-packages\opcua\ua\uatypes.py", line 231, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadInvalidArgument: One or more arguments are invalid.(BadInvalidArgument)

FIrst of all, it seems like "UaStatusCodeError(self.value)" is a result given by Kepware, correct? I have checked and rechecked certificates, username, password,etc..  But this error keeps popping up.

Some help would be appreciated. Thanks.

zerox1212, 2020-03-22:
Does it work with UA Exepert test client? FYI Kepware server is known to be bad.

yanndeferne, 2020-03-22:
Yes, I have tried the following "configurations" all with Basic256Sha256:
- Integration object's opc ua client to Kepware server : OK
- UAexpert test client to Kepware server : OK (as you suggested)

- Python opc ua client to Prosys OPC ua simulation server: OK
- Python opc ua client to Kepware server : not OK (see above error),
- Python opc ua client to UAexpert server : not OK

I believe that this issue does not come from the certificates.. But at this point I find it hard to understand what the source of the problem is... I tried to look into the code but if it's the server that refuses connection then I am not sure what else I can try.

What is bottering me is that it does work with other clients (integration objects and UA expert)..... And that Prosys server accepted connection using python opc ua client.

GreenHorn168, 2020-12-04:
@yanndeferne I've been struggling to create an OPC UA client with python's freeopcua and the Prosys UA server simulator. I can create the client without using the certificates and keys without any issues, but when I try and include the security related code (using the certificates and keys that come with the Prosys simulator) into my program it throws out the error message "Password was not given but private key is encrypted". As I understand it this means that the key (i.e. .pem file) is encrypted. Opening the .pem file in a text editor confirms this. How did you get around this? The set_security_string does can't accept a password for the key file. Or did you somehow replace the .pem file used by Prosys with one that wasn't encrypted?

I'd appreciate any advice you could give. I didn't want to copy my code into this tread as it's not related to the thread topic. If possible, could we have this discussion in a new thread?