FreeOpcUa/python-opcua #980: Modeling events in UAModeler, import them via xml, firing/cathing them seems strange

Splashcom, 2020-03-25:
Hey all, 

this one is difficult to explain: While using UAModeler for modeling Events i run into a strange behavior. Importing the model via xml allows me to browse the events, fire them, catch them, see self modeled properties or siblings. Seems working as expected.

One test case checks whether an event is fired after doing some magic in the namespace. The event should pass some data like nodeids, etc as properties. before triggering the event all data is regular stored in the properties (checked it via debugger). Catching this event in my test suit shows: The properties are now empty (None). Damn!

When i ignore my MDD approach and add the event properties on server start myself
```python
event.add_property(self.__ns_dialog, 'Type', ua.Variant(0, ua.VariantType.String))
```
the same test case passes. The properties of the captured event are now set correctly. To my regret i cannot provide a running example showing this strange behavior. it's deeply nested... 

I see several possible sources of bugs:
+ I can't use the tools properly
+ UAModeler xml-export function works not as expected 
+ while XMLImport something fails
+ data get lost while firing and catching UAModeler modeled events
+ ?  
 
Using:
+ UAModeler: 1.62
+ python-opcua: 0.98.7
+ Ubuntu LTS 18.04

Maybe you run in the same issue too or have some advice? I don't look for a solution or bugfix. Just want to ask the devs and start (maybe) a discussion.

oroulet, 2020-03-25:
Thanks for the feedback. I use that software quite a lot at work, and I was expecting more people would be interersted to contribute but it has not be the case so far...
>> Importing the model via xml allows me to browse the events, fire them, catch them, see self modeled properties or siblings
Great to know! Unfortunately I am modeling enums, structures and address space but in fact we are not using events, so I do not know much about that...

maybe you should try to model a similar event in unified automation modeller and test it /compare xml, maybe you will find something..

oroulet, 2020-03-25:
oops I see I misread the bug report. I thought you were using the opcuamodeler, while you are obviously using the modeler from unifired automation .....

swamper123, 2020-03-30:
To cheer you up @oroulet I love and use the modeler. Could you may add a list what works/shall work like for the server? This may motivate some people to add features.
@Splashcom It is difficult to follow, if there isn't a minimal example, so I (and I guess others as well) can just guess: How does your Event check the nodes? Maybe it could be caused by your magic.

oroulet, 2020-03-30:
nice to here @swamper123  ;-)

oroulet, 2020-03-30:
the main issue with modeller is probably not the missing feature but all the small bugs around because it is not well tested and I do not receive so many patch on it

Splashcom, 2020-03-31:
@swamper123 you're completely right. 

This one handles the event: Set properties and trigger it. 
```python    
def handle_event(self, event_instance, node: Node, instance_removed: bool):

        event_instance.event.NodeId = node.nodeid
        event_instance.event.InstanceRemoved = instance_removed
        feature_selection = node.get_child('1:selection').get_value()
        if feature_selection == 1:
            event_instance.event.Type = 'PositiveAttributeType'
        elif feature_selection == 2:
            event_instance.event.Type = 'NegativeAttributeType'
        else:
            event_instance.event.Type = ''
        # I'm firing my laser
        event_instance.trigger()
```

Event model (xml-export via UAModeler):

```xml
    <UAObjectType NodeId="ns=1;i=1019" BrowseName="1:ExampleEvent">
        <DisplayName Locale="en">ExampleEvent</DisplayName>
        <Description Locale="en"></Description>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=2041</Reference>
            <Reference ReferenceType="HasProperty">ns=1;i=6142</Reference>
            <Reference ReferenceType="HasProperty">ns=1;i=6144</Reference>
            <Reference ReferenceType="HasProperty">ns=1;i=6143</Reference>
        </References>
    </UAObjectType>
    <UAVariable DataType="NodeId" ParentNodeId="ns=1;i=1019" NodeId="ns=1;i=6142" BrowseName="1:NodeId" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>NodeId</DisplayName>
        <References>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=1019</Reference>
        </References>
    </UAVariable>
    <UAVariable DataType="Boolean" ParentNodeId="ns=1;i=1019" NodeId="ns=1;i=6144" BrowseName="1:InstanceRemoved" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>InstanceRemoved</DisplayName>
        <References>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=1019</Reference>
        </References>
    </UAVariable>
    <UAVariable DataType="String" ParentNodeId="ns=1;i=1019" NodeId="ns=1;i=6143" BrowseName="1:Type" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>Type</DisplayName>
        <References>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=1019</Reference>
        </References>
    </UAVariable>
```

Here the test case :

```python
    def test_adding_a_feature_fires_event(self) -> None:

        return_value_method_add_feature = '1234'
        parent_node_id = self.__spawn_dialog_node_in_namespace()
        node_manager = self.__node_manager

        class TestAddingAFeatureFiresEventSubHandler(object):
            test_still_running = True
            event = None

            def event_notification(self,
                                   _event) -> None:
                self.test_still_running = False
                self.event = _event
                return None

        def sub_thread():
            nonlocal return_value_method_add_feature, parent_node_id, event_emitter_node, node_manager
            return_value_method_add_feature = node_manager.call_method('1:addFeature', parent_node_id, event_emitter_node.nodeid, event_emitter_node.nodeid, event_emitter_node.nodeid, 1)

        event_emitter_node = self.__client.get_node(parent_node_id).get_child(['1:ExampleEmitter', '1:Attributes'])
        node_id_event = self.__client.get_root_node().get_child(['0:Types', 0:EventTypes', 0:BaseEventType', '1:ExampleEvent'])

        sub_handler = TestAddingAFeatureFiresEventSubHandler()
        sub = self.__client.create_subscription(100, sub_handler)
        handle = sub.subscribe_events(event_emitter_node, node_id_event)

        _thread = threading.Thread(name='test_adding_a_feature_fires_event: selected',
                                   target=sub_thread)
        _thread.start()

        while sub_handler.test_still_running:
            pass

        assert sub_handler.event.EventType == node_id_event.nodeid
        assert sub_handler.event.NodeId == return_value_method_add_feature
        assert sub_handler.event.InstanceRemoved is False
        assert sub_handler.event.Type == 'PositiveAttributeType'

        # clean up
        sub.unsubscribe(handle)
        sub.delete()
        return None
```
This test runs successfully if I have to perform the following mapping manually when starting the server after reading the namespace models.

```python
        example_event.add_property(1, 'NodeId', ua.Variant(0, ua.VariantType.NodeId))
        example_event.add_property(1, 'InstanceRemoved', ua.Variant(0, ua.VariantType.Boolean))
        example_event.add_property(1, 'Type', ua.Variant(0, ua.VariantType.String))
```
Without manual assignment the software test fails.