FreeOpcUa/python-opcua #981: OPCUA server.stop() crashes when running as a windows service

dpeterson3, 2020-03-25:
I am building a DA UA converter that runs as a windows service. When I stop the service, I get a crash report in the logs ending with asyncio\proactor_events.py: Value error: set_wakeup_fd only works in the main thread. This appears to be the same issue as 

https://stackoverflow.com/questions/60359157/valueerror-set-wakeup-fd-only-works-in-main-thread-on-apache2-4-41-python-3-8

Which is related to Python 3.8. Adding the lines
`
if sys.platform == "win32" and sys.version_info >= (3, 8, 0):
      asyncio.set_event_loop_policy(asyncio.WindowsSelectorEventLoopPolicy())
`
to the opcua.server.binary_server_asyncio.py file directly below the imports and before the OPCUAProtocol class fixes the issue as suggested in the above link.

oroulet, 2020-03-28:
Why is that stuff necessary. One of the main point of python is write code that is os independent. This really look like a Python bug. Is this going to be fixed? Or should we do something differently?

swamper123, 2020-03-30:
This problem is a discussed [Bug in Python](https://bugs.python.org/issue37373). I would suggest to not make a workaround by ourselves, but shall list this incompatibility Problem anywhere (for future users) until there is an official solution for that.

dpeterson3, 2020-03-30:
I agree that this is not a bug with OPC UA itself, and that it is a problem the asyncio maintainers need to fix. However, I do not know how long it will be before they fix this problem, and wanted to document a work-around here until that happens. I also wanted to make the maintainers here aware of the problem in case someone else runs into it.

oroulet, 2020-03-30:
thank you for the report @dpeterson3 and the explanation @swamper123 !