FreeOpcUa/python-opcua #985: XML export can not build etree

sntspyridon, 2020-03-30:
Hello,

I am trying to export a server adressSpace, and when i am exporting, this error occuress on multiple nodes and it does not write them to xml file.

WARNING:opcua.common.xmlexporter:Error building etree for node Node(StringNodeId(ns=4;s=MotionDeviceSystem)): The attribute is not supported for the specified Node.(BadAttributeIdInvalid)

Can you help me please ?

swamper123, 2020-03-31:
Have you debugged or just executed stuff?
This would help to find your problem.

sntspyridon, 2020-03-31:
I have just executed python client with xmlexporter.
This is what i see in the browser with the bad nodes.
[image]

zerox1212, 2020-03-31:
How are you creating these objects to get these errors? These properties should definitely work by default. The exporter isn't working because it expects these to be valid and I'm not sure we should just ignore them as you say.

If we ever imported the XML again what would we do with the blank fields? We would have to add a bunch of extra checks and if there is no data fill in some default.

I think the problem is not with the exporter, but with your address space.

sntspyridon, 2020-03-31:
I didnt create the objects in the server i am just trying to copy the server, to simulate the server localy.

Can i somehow repair the AdressSpace of the server ?
Or can i still somehow export the nodes and set the descriptions to something ?

zerox1212, 2020-03-31:
Fixing the server requires investigating what is wrong with those items. Could be difficult to do unless you can set break points and inspect the objects.

Having the exporter could be done but it will be difficult. The XML stuff is very complex with a lot of recursive functions. You would need to customize the exporter to ignore these bad nodes when it tries to serialize them.

sntspyridon, 2020-04-02:
The problem is that, it raises an exception for every description that is missing, so i changed a definition in the opcua.common....node.py file

the def get_attribute(self, attr):
result[0].statuscode.isgood() <--- this line

isgood() will not raise an exception.

zerox1212, 2020-04-02:
Glad you got it working. You should still fix your address space though. :)