FreeOpcUa/python-opcua #987: Best way to instantiate optional ObjectType children

janfelixklein, 2020-03-31:
Hello everyone,

I have a question on your implementation regarding the instantiation of an object based on a ObjectType which has some optional Methods and Variables.

As I see, there exist two variants for instatiating objects:

`Object = objects.add_object(nodeid, bname, objecttype=None)`
which will not resprect any optional children

or

`instantiate(parent, node_type, nodeid=None, bname=None, dname=None, idx=0, instantiate_optional=True)`

which will instantiate all optional children.

But what would be the recommended way, if I only want some of the children to instantiated?

zerox1212, 2020-03-31:
Can you give us an example of your object type, and then the output of what you would like instantiated?

oroulet, 2020-03-31:
if you only want to instantiate some childrens then you need to look at code and make your own method. It is probably very hard to implement in a generic way