FreeOpcUa/python-opcua #988: Decode Extension Object as python method input

janfelixklein, 2020-04-01:
Hello everyone,

maybe it's a stupid question, but I was searching all along for the right answer and wasn't successful, so I decided to bother you guys here.

I have a method specified in my NodeSet, which takes as input a customn Datatype (which holds 3 double values) and currently outputs a boolean value.

In my server file, i use the `link_method()` function to link it to a python function specified which currently looks like the following:

```
def drive_to_position(parent, goalPosition):
    # decode the input and do something with it
    return True
```

What I haven't figured out yet is how to decode the input which is of the following type:
`Variant(val:ExtensionObject(TypeId:FourByteNodeId(ns=2;i=5002), Encoding:1, 24 bytes),type:VariantType.ExtensionObject)`

What functions should be preferably used to decode the Extension object or what would be the right way? I guess I somehow need to link to the structure definition using the TypeId?

janfelixklein, 2020-04-02:
Well, i found what i was missing.
On server side I forgot to load the type definitions, which made the automatic decoding to not work.

`self.server.load_type_definitions()`

closed 2020-04-02 by janfelixklein

ArkasDev, 2020-11-03:
@janfelixklein Could you provide a code sample? #1157