FreeOpcUa/python-opcua #993: How to connect to variables given as paths

yenicelik, 2020-04-11:
I am trying to connect to variables in an OPC server. I have written an application which connects by specifying the path from the root in a comma-separated-path like this:

```
PATH_TO_NODE=0:Objects,2:Child1,2:NodeName
```

However, I just received details that I have to connect to the application using a "common" path like so:

```
Root\Objects\Child1//NodeName
```

or also reading from an array
```
Root\Objects\Child1//NodeArray/array[1]
```

How can I achieve this? I have not found any examples that show how this can be done.

oroulet, 2020-04-11:
for some historical reasons we use command and not slash as seperator, so just change one to another. But the paths you give as example are missing the namespace so they are not correct

yenicelik, 2020-04-11:
So together with the namespaces, would this correspond to something like the following?

```
0:Root\0:Objects\2:Child1//2:NodeName
```

Sorry if this is a stupid question, just new to OPC

oroulet, 2020-04-11:
yes change backslash to comma, add  and put is as arg to root.get_child()

I remember there was a dicusssion that we should change comma with backslash but I cannot find it again...