FreeOpcUa/python-opcua #995: How to use OPCUA Client with nsuri

sasauz, 2020-04-15:
Hallo,

I'm trying to subscribe to OPC-UA node with nsuri. It is loock licke:
ns=3;s=OPC_UA.nsuri=http://opcuatest.com/something/;s=level1.level2.l3.l4

My Quell-Code looks like:

> var = client.get_node("ns=3;s=OPC_UA.nsuri=http://opcuatest.com/something/;s=level1.level2.l3.l4")
 print(var)
 v = var.get_value()
 print(v)

This is the output
> Node(StringNodeId(ns=3;s=level1.level2.l3.l4))
Traceback (most recent call last):
  File "opcuaclient.py", line 43, in <module>
    v = var.get_value() # get value of node as a python builtin
  File "/home/user/.local/lib/python2.7/site-packages/opcua/common/node.py", line 153, in get_value
    result = self.get_data_value()
  File "/home/user/.local/lib/python2.7/site-packages/opcua/common/node.py", line 162, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
  File "/home/user/.local/lib/python2.7/site-packages/opcua/common/node.py", line 274, in get_attribute
    result[0].StatusCode.check()
  File "/home/user/.local/lib/python2.7/site-packages/opcua/ua/uatypes.py", line 231, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadNodeIdInvalid: The syntax of the node id is not valid.(BadNodeIdInvalid)

get_node-Method loose "s=OPC_UA.nsuri=http://opcuatest.com/something/".

Can some one help me?

AndreasHeine, 2020-04-21:
According to: http://documentation.unified-automation.com/uasdkhp/1.0.0/html/_l2_ua_node_ids.html

What is the "Identifier Type" of this variable you want to subscribe?

sasauz, 2020-04-22:
It is String:
[image]

AndreasHeine, 2020-04-22:
from above:
var = client.get_node("ns=3;s=OPC_UA.nsuri=http://opcuatest.com/something/;s=level1.level2.l3.l4")

have you tried to use:
var = client.get_node("ns=3;s=level1.level2.l3.l4")

it looks strange to me if i see two times "s=" => "ns=3;s=xxx;s=yyy"

this looks like a namespaceuri: http://opcuatest.com/something/
and this like the identifier: level1.level2.l3.l4

what system is it i have never seen such strange node id... sorry

sasauz, 2020-04-23:
> "s=" => "ns=3;s=xxx;s=yyy"
it is also new to me. It's look like a namespace uri with identifier inside of identifier

But what is really interesting, it is possible to read data with Softing OPC-UA Client, but not with other Clients/Libraries

> have you tried to use:
var = client.get_node("ns=3;s=level1.level2.l3.l4")

yes, I also try so, but it also doesn't work

oroulet, 2020-04-23:
Our library crashes for sure due to the extra ; inside your string urn. You will need to prove this is acceptable by spec before changing this

fcaspani01, 2022-12-16:
Is there any update??

I have the same problem but I get the value by navigate the nodes with get_children function but it doesn't work with get_node function.

[image]

AndreasHeine, 2022-12-16:
this is not a valid nodeid!

AndreasHeine, 2022-12-16:
@fcaspani01 

> 

> `ns=<namespaceIndex>;<identifiertype>=<identifier>`

see https://documentation.unified-automation.com/uasdkhp/1.4.1/html/_l2_ua_node_ids.html

AndreasHeine, 2022-12-16:
there are cases of an expanded nodeids

> `nsu=<namespaceUri>;<identifiertype>=<identifier>`

for which you need to read the namespacearray from server and resolve to

> `ns=<namespaceIndex>;<identifiertype>=<identifier>`