FreeOpcUa/python-opcua #997: Can't Set Value

PauloSilva98, 2020-04-16:
Can't seem to use the function set_value to change the value.
This is my client code: 
if __name__ == "__main__":

    client = Client("opc.tcp://DESKTOP-AFIK9N6:4840")
    # client = Client("opc.tcp://admin@localhost:4840/freeopcua/server/") #connect using a user
    try:
        client.connect()
        print("WE IN THIS BRUH")
        root = client.get_root_node()
        print("Objects node is: ", root)
        print("Children of root are: ", root.get_children())
        objects = client.get_objects_node()
        print("Objects node is: ", objects)
        print("Children of root are: ", objects.get_children())
        var = client.get_node("ns=4;s=|var|CODESYS Control Win V3 x64.Application.OPC_TEST.state")
        print(var.get_data_type)
        value = var.get_value() # get value of node as a DataValue object
        print(value)

        var.set_data_value(1)
        #print (value) 

        #Create a subscription
        
        handler = SubHandler()
        sub = client.create_subscription(500, handler)
        print("SUBSCRIPTION")
        handle = sub.subscribe_data_change(var)
        value = var.get_value()
    
I am accessing a server created by codesys and when i try to change the value of a INT a variable i get this error:
opcua.ua.uaerrors._auto.BadTypeMismatch: The value supplied for the attribute is not of the same type as the attribute"s value.(BadTypeMismatch)

AndreasHeine, 2020-04-16:
The error tells u everything! 

var.set_data_value(1) => 1 is a 64bit int in python ...

you need to create a datavalue object with opcua type and value:

    servicelevel_node = server.get_node("ns=0;i=2267") #Service-Level Node
    servicelevel_value = 255 #0-255 [Byte]
    servicelevel_dv = ua.DataValue(ua.Variant(servicelevel_value, ua.VariantType.Byte))
    servicelevel_node.set_value(servicelevel_dv)

PauloSilva98, 2020-04-17:
Thank you, it's fixed! Sorry about the ignorance, not very familiar with python.

closed 2020-04-17 by PauloSilva98

AndreasHeine, 2020-04-17:
No problem you're welcome :)

AndreasHeine, 2020-04-17:
in case of opc ua keep in mind that python types and opc ua types are different and a opc ua variable (datavalue object) has more information than just the raw value (timestamp, typeinfo ...)

zerox1212, 2020-04-19:
I strongly recommend you use a good IDE like PyCharm so you can pause execution and inspect the objects. OPC UA has many very complex objects that require debugging to understand.