FreeOpcUa/python-opcua #999: Some Clients Can't See FreeOpcUa Server

SackLunch, 2020-04-21:
Hello,

I am working on setting up a Python OPC UA Server and I am able to successfully work with clients such as UAExpert, and Prosys with no troubles.   The client I am trying to use, SAP PCo and also Quick Connect, can't seem to find the server though.  They are able to find Makitron's simulation server and National Instruments OPC Server.  I was looking into registering the server with a local discovery service and haven't had any luck yet with the Python.

Any suggestions if I'm on the correct path?  I guess I don't know why UAExpert can find the endpoint while my other clients can't.

Thank you for any help!

AndreasHeine, 2020-04-21:
Could you provide the opc ua related sourcecode just to check setup is ok?

SackLunch, 2020-04-21:
Just using a very basic server setup from a youtube setup.  Still learning and connecting dots so I'm hoping that I'm overlooking something simple.

```
import opcua
from opcua import Server
from opcua import Client
from random import randint
import datetime
import time

if __name__ == "__main__":

    server = Server()

    url = "opc.tcp://localhost:9999"
    server.set_endpoint(url)
    server.set_application_uri("urn:freeopcua:python:discovery")

    print(server.get_endpoints())
    print(server.set_server_name("Test Server"))

    name = "OPCUA_SIMULATION_SERVER"
    addspace = server.register_namespace(name)

    node = server.get_objects_node()

    Param = node.add_object(addspace, "Parameters")

    Temp = Param.add_variable(addspace, "Temperature", 0)
    Press = Param.add_variable(addspace, "Pressure", 0)
    Time = Param.add_variable(addspace, "Time", 0)

    Temp.set_writable()
    Press.set_writable()
    Time.set_writable()

    server.start()
    print("Server startet at {}".format(url))
    print(server.get_endpoints())

    while True:
        Temperature = randint(10, 50)
        Pressure = randint(100, 500)
        TIME = datetime.datetime.now()

        print(Temperature, Pressure, TIME)

        Temp.set_value(Temperature)
        Press.set_value(Pressure)
        Time.set_value(TIME)

        time.sleep(5)
```

AndreasHeine, 2020-04-22:
i am not sure that localhost is reachable for other clients outside your pc!

oroulet, 2020-04-22:
Yes that client is only reachable from local PC. Use * or a real ip address

AndreasHeine, 2020-04-22:
```
    def set_endpoint(self, url):
        self.endpoint = urlparse(url)
```

I am not that deep in urlparse... what does * do? is it like listen on all ifaces?

oroulet, 2020-04-22:
yes url = "opc.tcp://*:9999" will make program listen on all interfaces. Although  "opc.tcp://0.0.0.0:9999" might be more correct

SackLunch, 2020-04-22:
Hmm Still having some trouble.  Using  "opc.tcp://0.0.0.0:9999" I'm having trouble seeing it with UAExpert's client unless I set the endpoint in UAExpert as "opc.tcp://localhost:9999" which confuses me.  

Trying "opc.tcp://*:9999" I'm seeing a socket error :
"socket.gaierror: [Errno 11001] getaddrinfo failed"

Both my client and server are local so I'm not sure why I wouldn't be able to connect them.  I looked in the Windows registry and I can't see any service for this running either but I'm not sure if I'm supposed to.

oroulet, 2020-04-22:
0.0.0.0 and * are for server only. Use real ip from client

AndreasHeine, 2020-04-22:
and your real lan port adress? just for testing! without firewall enabled?

AndreasHeine, 2020-04-22:
ok ive tested some... it looks like uaexpert has trouble to find the ua server if it has a different port then 4840...

```
import opcua
from opcua import Server
from opcua import Client
from random import randint
import datetime
import time

if __name__ == "__main__":

    server = Server()

    url = "opc.tcp://localhost:4840"
    server.set_endpoint(url)
    server.set_application_uri("urn:freeopcua:python:discovery")

    print(server.get_endpoints())
    print(server.set_server_name("Test Server"))

    name = "OPCUA_SIMULATION_SERVER"
    addspace = server.register_namespace(name)

    node = server.get_objects_node()

    Param = node.add_object(addspace, "Parameters")

    Temp = Param.add_variable(addspace, "Temperature", 0)
    Press = Param.add_variable(addspace, "Pressure", 0)
    Time = Param.add_variable(addspace, "Time", 0)

    Temp.set_writable()
    Press.set_writable()
    Time.set_writable()

    server.start()
    print("Server startet at {}".format(url))
    print(server.get_endpoints())

    while True:
        Temperature = randint(10, 50)
        Pressure = randint(100, 500)
        TIME = datetime.datetime.now()

        print(Temperature, Pressure, TIME)

        Temp.set_value(Temperature)
        Press.set_value(Pressure)
        Time.set_value(TIME)

        time.sleep(5)
```

AndreasHeine, 2020-04-22:
[image]

SackLunch, 2020-04-22:
Ha, you guys are much faster than me!  I think sorting out what I was using for my client values did the trick.  I can test the connection to the server with my SAP PCo client now.

For reference if others run into this or when I inevitably forget:
I used my personal IP address as the Discovery Server value and then the Server Endpoint is what I set up in Python.

[image]

closed 2020-04-22 by SackLunch