FreeOpcUa/python-opcua #1001: How to connect ExtensionObjects and have access to all variables?

CanK93, 2020-04-22:
Hi guys,

I have a problem accessing all the data points in a extension object. E.g. I have an ExtensionObject as a Node which has three other values (speed1, x, y). How can I access them with this client?

Regards
Can

AndreasHeine, 2020-04-23:
> Furthermore, how it is possible to write new data? I often get the error "The server does not support writing the combination of value, status and timestamps provided.(BadWriteNotSupported)"

you can try:
`random_node.set_value(ua.DataValue(ua.Variant(random.randint(70,90), ua.VariantType.UInt32)))`

is the variable writable (serverside)?

CanK93, 2020-04-23:
Thank you for your answer Andreas. 
The variable should be writable. E.g. I can change the ExtensionObject.speed in UA Expert and write the new value.

oroulet, 2020-04-23:
did you remember to call load_type_definitions() ?

zerox1212, 2020-04-23:
Is there any way to raise a warning when access extension object and `load_type_definitions()` hasn't been called?