FreeOpcUa/python-opcua #1004: Kepware Bad Quality

AndrewStuhr, 2020-04-24:
I'm trying to read multiple Kepware tags at once.  If one or more tags has bad quality, the entire read fails.  Is there a way to get a bad quality status back for bad tags without the entire read failing?  Here's part of my code where I'm reading multiple tags at once:

nodeid 			= ua.NodeId.from_string(node)
attr 				= ua.ReadValueId()
attr.NodeId 		= nodeid
attr.AttributeId 	= ua.AttributeIds.Value
params.NodesToRead.append(attr)
result = client.uaclient.read(params)

Thanks

AndreasHeine, 2020-04-24:
what do you mean with failing is it a python error or a opcua statuscode?

AndrewStuhr, 2020-04-24:
It's an opcua status code :  bad 0x80000000.  

It takes about 3-4 seconds to from when the read request begins until opcua generates this statuscode.  I'm looking for a way to read multiple tags every second and get a bad quality response just for bad quality tags, yet still get tag values for good quality tags.  Any thoughts?

Thanks again!

AndreasHeine, 2020-04-24:
sounds impossible... one read request and multiple responses... if it would be an python server i would use a opcua-method or if the datatypes are equal just a variable with n dimensions... how many variables u plan to read?

AndrewStuhr, 2020-04-24:
I worded that poorly... I'm looking to get one response back with an array/object that contains read values for each node.  However, if a node had bad quality, the array/object would have some indication of the bad quality instead of the node's value.

Currently I was just reading a couple of nodes, but eventually I'm hoping to read a lot - around 1000 per second.

AndreasHeine, 2020-04-24:
ive read 2048 boolean io from a s7-1500 but it was only possible as an array (variable type boolean with arraydimension of 2048)! 

but to be honest with you ive never worked with kepware ive wrote the services by myself... 

what i wonder atm is why a node could have bad quality?

AndrewStuhr, 2020-04-24:
In this case, it's because Kepware is reading data from scale that may or may not have an object on it.  If there isn't an object on the scale, it responds to Kepware in a way that makes the Kepware quality go bad.  Once an item is being weighed, the quality in Kepware temporarily becomes good.

Do you think the current method I'm using to read multiple variables will be unable to handle 1000 nodes, even with all good quality?

AndreasHeine, 2020-04-24:
the thing is, it depends on the server hardware, network traffic, and a bunch of more things like opcua server config... u can read 1000 nodes for sure but in 1s interval...

python-opcua/opcua/client/ua_client.py:
```
    def read(self, parameters):
        self.logger.info("read")
        request = ua.ReadRequest()
        request.Parameters = parameters
        data = self._uasocket.send_request(request)
        response = struct_from_binary(ua.ReadResponse, data)
        self.logger.debug(response)
        response.ResponseHeader.ServiceResult.check()
        # cast to Enum attributes that need to
        for idx, rv in enumerate(parameters.NodesToRead):
            if rv.AttributeId == ua.AttributeIds.NodeClass:
                dv = response.Results[idx]
                if dv.StatusCode.is_good():
                    dv.Value.Value = ua.NodeClass(dv.Value.Value)
            elif rv.AttributeId == ua.AttributeIds.ValueRank:
                dv = response.Results[idx]
                if dv.StatusCode.is_good() and dv.Value.Value in (-3, -2, -1, 0, 1, 2, 3, 4):
                    dv.Value.Value = ua.ValueRank(dv.Value.Value)
        return response.Results
```

i assume your code looks like this:
```
for node in nodeList:
	nodeid = ua.NodeId.from_string(node)
	attr = ua.ReadValueId()
	attr.NodeId = nodeid
	attr.AttributeId = ua.AttributeIds.Value
	params.NodesToRead.append(attr)
result = client.uaclient.read(params)
```
if the nodelist does not change at runtime:
the first thing i would try is to run the forloop just once an store "params", so you gain some speed (1000 nodes => 1000 iterations to get params + 1000 iterations in the for loop of the read-method => 2000 iteration total ...) the next thing is to try multiprocessing/multithreading with more then one opcua-client-connection so u can split up work depends on your hardware... 

but all that does not help with the bad quality in kepware... :(

AndrewStuhr, 2020-04-24:
Thanks for the performance tips.  I'll try these out.

I was thinking I could modify the opcua library to handle the bad status code differently, but I don't think that would even help because it take's 3-4 seconds to just get the response.  Let me know if you have any other thoughts come to mind!  Thanks

zerox1212, 2020-04-24:
This sounds like a bug. I do not think an entire read request should fail due to one node having bad status.

However, it will be many times faster to subscribe to 1000 items then do an explicit read. Not to mention subscribing to 1000s nodes will solve your bad quality issue because one bad node won't stop the subscription from updating all the other ones in the subscription.

AndrewStuhr, 2020-04-24:
Damn, I haven't even looked in subscriptions.  I've just been reading on an interval.  I'm definitely try subscriptions, thank!

AndreasHeine, 2020-04-24:
@zerox1212: is the subscription on servers with encryption still broken or fixed? (OPC UA Server: AttributeError: 'NoneType' object has no attribute 'encrypted_block_size' #956)

zerox1212, 2020-04-24:
I don't know. I never used encryption when I used this library.

AndreasHeine, 2020-04-24:
python-opcua/examples/client_to_kepware.py:
```
        handler = SubHandler()
        sub = client.create_subscription(500, handler)
```
like:
```
for node in nodeList:
    handleList = sub.subscribe_data_change(client.get_node(node))
```

AndrewStuhr, 2020-04-28:
Thanks for the code Andreas!  I'm able to set up a subscription, and it's handling bad quality tags (it just returns 'None').

One final question: I'm running your script in Window's command prompt.  Is there a way to exit the script?  Currently, CTL + C isn't doing anything.  Is there something I have to do with client.disconnect()?

Thanks again

AndreasHeine, 2020-04-28:
try:
        #subscribe loop
except KeyboardInterrupt:
        #unsubscribe loop
        #disconnect

AndrewStuhr, 2020-04-28:
I used the following code and now I can stop the script with CTL + C - thanks. 

The only issue is I get 2 output messgaes:
"ServiceFault from server received while waiting for publish response" (twice)

These seem more like print statements than errors because there's no tracebacks.  Do you know if I'm doing something wrong?

```
if __name__ == "__main__":
    try:
        client = Client("opc.tcp://localhost:49320")
        client.connect()

        handler = SubHandler()
        sub = client.create_subscription(500, handler)
        handleList = []
        for node in nodeList:
            handleList.append(sub.subscribe_data_change(client.get_node(node)))
			
    except:
        client.disconnect()
		
    try:
        while 1:
            time.sleep(0.5)
			
    except KeyboardInterrupt:
        for handle in handleList:
            sub.unsubscribe(handle)
			
        sub.delete()
        client.disconnect()
```

AndreasHeine, 2020-04-28:
I actually had the same issue with the double printed it appearce that there is a bug... but not with pub/sub i got it with opcua-method-call... till now i could not find anything...

AndrewStuhr, 2020-04-28:
Interesting... please let me know if you figure it out.  And thanks again, you rock man!