FreeOpcUa/python-opcua #1005: Python minimal example: Reading the value of a variable

LosWochos76, 2020-04-24:
Hello everybody,

i'm just starting to work with the python version of FreeOpcUa by checking the examples.
Unfortunately, I directly run into a problem that I do not understand.

The server-minimal.py contains the following lines. In my understanding a new variable with the value 6.7 is created and added to the servers adress space.

   objects = server.get_objects_node()
   myobj = objects.add_object(idx, "MyObject")
   myvar = myobj.add_variable(idx, "MyVariable", 6.7) 

The client-minimal.py then tries to read the value of that exact variable:

   myvar = root.get_child(["0:Objects", "2:MyObject", "2:MyVariable"])
   print("Value: ", myvar.get_value())

Unfortunately, the result is everything but 6.7. It seems to be some random value.
Furthermore, everytime I start the client the value changes to something else.
What is happening here? Is this a bug?

Best regards,

Alex

AndreasHeine, 2020-04-24:
Hey,

> The server-minimal.py contains the following lines. In my understanding a new variable with the value 6.7 is created and added to the servers adress space.
> 
> objects = server.get_objects_node()
> myobj = objects.add_object(idx, "MyObject")
> myvar = myobj.add_variable(idx, "MyVariable", 6.7)

correct!

but in line 159 its got overwritten:
`server.set_attribute_value(myvar.nodeid, ua.DataValue(9.9))  # Server side write method which is a but faster than using set_value`

[image]

LosWochos76, 2020-04-24:
Oh my! Havent seen that. Thank you for your help!

AndreasHeine, 2020-04-24:
your welcome ;)