FreeOpcUa/python-opcua #1006: server-example.py

AndreasHeine, 2020-04-24:
Hey,

looks like something went wrong with the latest update on guid node id in the server.example.py:
[image]

python 3.7.3 32bit
opcua 0.98.9

AndreasHeine, 2020-04-24:
@Michael-List

zerox1212, 2020-04-24:
There was some work on GUID recently. I guess `NodeId()` expects a string instead of a UUID object. Probably needs PR and review as we decided the to use UUID objects in python and never GUID as a string or bytes.

Michael-List, 2020-04-25:
@AndreasHeine I will take a look at it

Michael-List, 2020-04-25:
@AndreasHeine I could identify the problem.

- Version 0.98.9 was released on 18.12.19.
- On 14.03.20 I spotted a bug with guid node id that was fixed in this pull request: https://github.com/FreeOpcUa/python-opcua/pull/975/files
- In this pull request I also added a guid node id example to the server-example.py because there was none.

If you are now using the newest server-example.py file with the lib version 0.98.9 my pull request is missing. Thats why this error is happening. Either you comment out my guid node id example if you don't need it or you install the python lib directly from github so that you are getting the newest version.

AndreasHeine, 2020-04-25:
ok ty :)

zerox1212, 2020-04-26:
@oroulet Maybe should do a new release.

closed 2020-06-01 by AndreasHeine