FreeOpcUa/python-opcua #1010: Python OPC UA server with Matrikon Tunneller

dns28, 2020-04-28:
Hello everyone,

I have an OPC UA server running fine using this python library. I can access the tags using UaExpert without problems.

Now I need to use Matrikon OPC Tunneller to be able to get data from a OPC DA Client. The tunneller can establish the connection to the OPC UA server but can't create subscriptions. I asked help to Matrikon and they tried to solve the problem but they say that this python OPC UA library is not fully compliant.

The log file from tunneller is below:

[04/27 12:09:37.452](9,1776) CPSTServer::Run Service started. ProgId = [UAProxy.freeopcua_python_server.1]
[04/27 12:09:51.969](2,0832) CFOPCServer::CreateInstance Creating new instance of the OPC Server
[04/27 12:09:51.985](1,0832) Com2UaOpcServer::Init Initializing OPC Server object for the client : DESKTOP-PGTCHBV\siemens.
[04/27 12:09:51.985](1,0832) Com2UaSession::Connect Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#None
[04/27 12:09:51.985](1,0832) Com2UaSession::Connect Session User Identity Token Type: Anonymous
[04/27 12:09:52.047](1,0832) Com2UaSession::GetOperationLimits Vendor info: Matrikon International +1-780-945-4011 http://www.matrikonopc.com
[04/27 12:09:52.063](1,0832) Com2UaSession::GetOperationLimits Maximum Monitored Items Per Call: 4294967295
[04/27 12:09:52.078](1,0832) Com2UaSession::GetOperationLimits Maximum Nodes Per Browse Call: 4294967295
[04/27 12:09:52.094](1,0832) Com2UaSession::GetOperationLimits Maximum Nodes Per Read Call: 4294967295
[04/27 12:09:52.110](1,0832) Com2UaSession::GetOperationLimits Maximum Nodes Per Write Call: 4294967295
[04/27 12:09:52.110](1,0832) Com2UaSession::Connect Session: Matrikon OPC UA Proxy from OPC Client DESKTOP-PGTCHBV\siemens\1 created successfully. MaxRequestMessageSize: 65536, RequestedSessionTimeout: 1200000, RevisedSessionTimeout: 1200000
[04/27 12:09:52.110](1,0832) Com2UaBranch::Init Obtained server.
[04/27 12:09:52.110](1,0832) Com2UaOpcServer::CreateBrowser Initializing browser for client : DESKTOP-PGTCHBV\siemens
[04/27 12:09:52.110](1,0832) Com2UaBranch::Init Obtained browser.
[04/27 12:09:52.110](1,0832) Com2UaBranch::Init Rebrowsing.
[04/27 12:09:52.829](1,0832) Com2UaBranch::Init Rebrowsed.
[04/27 12:09:52.829](1,0832) CPSTServer::ClientUp CLIENT CONNECTION: DESKTOP-PGTCHBV\siemens, 1 clients connected.
[04/27 12:09:52.829](2,0832) CFOPCServer::CreateInstance Object creation result code is [0x0]
[04/27 12:09:55.985](1,4956) IsLicensed License state is changed from UNLICENSED to LICENSED.
[04/27 12:09:55.985](1,0832) Com2UaSubscription::CreateUaSubscription CreateSubscription successful for group: Group0. UA Server SubscriptionId: 2912, UA Client SubscriptionId: 18468, Requested Publishing Interval: 1000
[04/27 12:09:55.985](1,0832) ComOpcGroup2UaSubscription::CreateUaSubscription Subscription 2912 created successfully for group: Group0
[04/27 12:09:56.032](1,0832) Com2UaSubscription::SetPublishingMode SetPublishingMode failed for group: Group0. Error Message: 0x80400000: Requested operation is not implemented
[04/27 12:09:56.048](1,0832) Com2UaSubscription::ModifySubscription ModifySubscription failed for group: Group0 since UA server returned invalid publishing interval: 0.000000. Reseting publishing interval to existing value. Error Message: 0x0: Success
[04/27 12:09:56.063](1,0832) Com2UaSubscription::SetPublishingMode SetPublishingMode failed for group: Group0. Error Message: 0x80400000: Requested operation is not implemented
[04/27 12:09:56.079](1,0832) Com2UaSubscription::SetPublishingMode SetPublishingMode failed for group: Group0. Error Message: 0x80400000: Requested operation is not implemented
[04/27 12:09:56.095](1,0832) Com2UaSubscription::DeleteUaSubscription DeleteSubscriptions response result: 0x0: Success. Group Name: Group0
[04/27 12:09:56.095](1,0832) Com2UaSubscription::DeleteUaSubscription UA Subscription deleted for group: Group0

Has anyone experience working with this library with Matrikon OPC Tunneller?

Your help is much appreciated.

zerox1212, 2020-04-29:
Maybe the server doesn't have SetPublishingMode implemented. I'm not sure what this does; would need to check UA spec.

brubbel, 2020-04-30:
The fastest way to get help here is to provide an example, which describes the exact steps to reproduce the problem. In this case, the setup of the freeopcua server, matrikon tunneller, plus a dummy DA device to connect to.

dns28, 2020-05-01:
Hi everyone,

The code of the python OPC UA server is attached.
[opc_server.txt](https://github.com/FreeOpcUa/python-opcua/files/4563260/opc_server.txt)

Basically it takes data from a database and creates one OPC UA tag for each row of the results. The OPC UA server implementation is pretty much the server sample code provided in GitHub.

This works fine with UaExpert.
Do I need to add something to support subscriptions?

Many thanks